oxullo/Arduino-MAX30100 #3: MAX30100 and BLE Nano

AzraelNB, 2016-09-14:
I'm trying to get your code to compile on a BLE Nano from RedBearLabs. Unfortunately I keep running into this issue:
 
MAX30100.cpp:30:10: error: 'class TwoWire' has no member named 'setClock'

     Wire.setClock(I2C_BUS_SPEED);

          ^

exit status 1
Error compiling for board BLE Nano(V1.5 32KB).

I realise that the problem lies with the adapted wire library for the BLE Nano, but I was wondering if you had any thoughts on how I could address this problem?

oxullo, 2016-09-14:
@AzraelNB as far as I can see here: https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/libraries/Wire/Wire.cpp yes, there's no setClock() method. Check if 970e2d404c87b177beb398eaad9346b317eb68e9 works for you. I haven't tested it at all, though!

AzraelNB, 2016-09-15:
Thanks! Well, it resolved that issue, but there seem to be a whole number of other issues popping up now too:/ Seems getting it to work on the BLE Nano will take some work.

oxullo, 2016-09-16:
I imagined.. In case you see the chance for a PR, I'll be glad to review it.

closed 2016-09-16 by oxullo