oxullo/Arduino-MAX30100 #4: Undefined reference

dicobrazz, 2016-09-24:
Hi, thank you for the great code! 

I have an issue of undefined reference, it might just because I am a total beginner:)

Arduino: 1.6.12 (Mac OS X), Board: "Arduino/Genuino Uno"

/var/folders/r2/h34h074s3xv3wdfcgs8l5_yw0000gn/T//cc44E7VM.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_MAX30100_Minimal.ino.cpp.o.1945':
cc44E7VM.ltrans0.o:(.text.startup+0x74): undefined reference to `PulseOximeter::PulseOximeter()'
/var/folders/r2/h34h074s3xv3wdfcgs8l5_yw0000gn/T//cc44E7VM.ltrans0.ltrans.o: In function `main':
cc44E7VM.ltrans0.o:(.text.startup+0x170): undefined reference to `PulseOximeter::begin(PulseOximeterDebuggingMode)'
cc44E7VM.ltrans0.o:(.text.startup+0x17c): undefined reference to `PulseOximeter::setOnBeatDetectedCallback(void (*)())'
cc44E7VM.ltrans0.o:(.text.startup+0x184): undefined reference to `PulseOximeter::update()'
cc44E7VM.ltrans0.o:(.text.startup+0x1ce): undefined reference to `PulseOximeter::getHeartRate()'
cc44E7VM.ltrans0.o:(.text.startup+0x324): undefined reference to `PulseOximeter::getSpO2()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

oxullo, 2016-09-24:
Hey @dicobrazz it compiles fine under the same premises (arduino:1.6.12, board:Uno, max30100lib:1.0.0, OSX:10.11.6 example:MAX30100_Minimal). Could you post the code you're trying to compile? Have you tried to compile the examples?

dicobrazz, 2016-09-24:
thank you @oxullo for a prompt reply!
I am testing the example code "Minimal", after installing your library through "library manager"
Haven't changed a thing.

`
/*
Arduino-MAX30100 oximetry / heart rate integrated sensor library
Copyright (C) 2016  OXullo Intersecans <x@brainrapers.org>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include <Wire.h>
#include "MAX30100_PulseOximeter.h"

#define REPORTING_PERIOD_MS     1000

// PulseOximeter is the higher level interface to the sensor
// it offers:
//  * beat detection reporting
//  * heart rate calculation
//  * SpO2 (oxidation level) calculation
PulseOximeter pox;

uint32_t tsLastReport = 0;

// Callback (registered below) fired when a pulse is detected
void onBeatDetected()
{
    Serial.println("Beat!");
}

void setup()
{
    Serial.begin(115200);

    // Initialize the PulseOximeter instance and register a beat-detected callback
    pox.begin();
    pox.setOnBeatDetectedCallback(onBeatDetected);
}

void loop()
{
    // Make sure to call update as fast as possible
    pox.update();

    // Asynchronously dump heart rate and oxidation levels to the serial
    // For both, a value of 0 means "invalid"
    if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
        Serial.print("Heart rate:");
        Serial.print(pox.getHeartRate());
        Serial.print("bpm / SpO2:");
        Serial.print(pox.getSpO2());
        Serial.println("%");

        tsLastReport = millis();
    }
}`

dicobrazz, 2016-09-24:
I have no idea what it was, i just shut down the app and opened the project again and it worked!! I am so sorry!

dicobrazz, 2016-09-24:
Have you tested it with Arduino 101?

closed 2016-09-24 by dicobrazz

oxullo, 2016-09-24:
It requires minimal intervention to make it working:

1. add an include for <Wire.h> to the sketch
2. comment out MAX30100.cpp:30

bye!