oxullo/Arduino-MAX30100 #6: hello i want to ask you why my chip cannt work

jackccchen1, 2016-10-18:
i use a modle  ,but cannt work  what can i do

jackccchen1, 2016-10-18:
https://www.google.com.hk/search?q=max30100&prmd=ivmn&tbm=isch&tbo=u&source=univ&fir=TAEsiHUy7oz6NM%253A%252CGaNS6pPrbeq0SM%252C_%253BXukKVzs7hh7omM%253A%252CUSta48v5AdzA7M%252C_%253BGBuKZXYYhY1g6M%253A%252CL5s4ffbj0RQxEM%252C_%253B4UFtAnQl7DIVVM%253A%252CRKmfFlwx33FZ_M%252C_%253BNAe2ai7PxSFogM%253A%252CnL509GS6Iweu9M%252C_%253BkBDdcHCd-PnEBM%253A%252CUSta48v5AdzA7M%252C_%253B2xEz0WOtTMzGoM%253A%252CPTZW62gxqcCrLM%252C_%253BxIbkwCiKyhfD2M%253A%252CsUhkdFAbbcjp7M%252C_%253BD-XrJH1K999vVM%253A%252C3XSnsmuSuZ2pyM%252C_%253BmIcYqi_9TDDi4M%253A%252C3XSnsmuSuZ2pyM%252C_&usg=__8MKNdETaujAFgdpIgfJUE42_lTg%3D&sa=X&ved=0ahUKEwiulLaQ9-PPAhVLu48KHaUdAFgQ7AkIZw

jackccchen1, 2016-10-18:
like the 3nd

oxullo, 2016-11-05:
Hey @jackccchen1 I'd need more info to understand the issue you're facing. As long as you have a breakout which exposes the I2C interface of the MAX30100 you should be good to go. Check if the breakout include the mandatory I2C pullups.

closed 2016-11-13 by oxullo