oxullo/Arduino-MAX30100 #7: About the data reading

JulzDuf, 2016-10-28:
Hi! I've looked in your library and I see that you have the interrupt register progrmmed but never used in the library.

The way I understand your program, the arduino asks regurlarly for the data of the MAX30100 instead of asking when receiving an interrupt.

Am I misunderstanding the way it works? If not, do you have a solution to enable the use of those interrupts? 

Thanks in Advance!

oxullo, 2016-10-28:
Hey @JulzDuf , correct, the PulseOximeter component of the library is based on a polling strategy (the magic happens in PulseOximeter::checkSample()).
The interrupt enable register (addr=0x01) is left untouched, which means it retains its POR state (interrupt not enabled).
Handling the interrupt requires a bit more code and since I wanted the implementation of the PulseOximeter component to be minimal, I decided to use polling.

If you want to use an interrupt-driven approach:
1. make sure to connect the interrupt output from the MAX30100 to a µC pin that supports level-triggered interrupt (eg: INT0 on any AVR)
2. enable the interrupt sources relative to the data feeds you want to read (HR, SO2, TEP)
3. make sure your ISR checks for the interrupt status (0x00) register and reads the indicated register accordingly

Lot more is left to be described, in particular if you want to retain the high level functionalities of the PulseOximeter component (which is not really suitable for subclassing) or if you want to exploit the FIFO feature offered by the chip.

closed 2016-11-05 by oxullo

zahrahalwachi, 2017-03-05:
Hello @JulzDuf did you find a way to get the interrupts?

DallySon, 2017-03-05:
Dear Sir,

 

You did mistake with address. It is not my message:)