oxullo/Arduino-MAX30100 #9: Extracting Temperature from the MAX30100

slucero1124, 2016-11-08:
I didn't see any provisions in your library for extraction of the temperature measured by the sensor.  Is that correct, or am I missing it?  How might I go about adding that functionality?

Thanks,
Steven

oxullo, 2016-11-30:
Hey @slucero1124 , it's correct, there's currently no code to get a temperature readout from the sensor.
The tricky part is that the readout is asynchronous: a flag must be set and checked for it to be cleared before reading the temperature registers.
I don't have a breakout at hand at the moment, but I will for the weekend. I'll see if I have some time to implement the temperature readout, probably it's faster than hinting how to do it :)

What I'd do, keeping the current polling fashion, is to set the 0x06.TEMP_EN (page 15 of the DS) and poll it until it clears. At this point the temperature can be assembled using the two registers 0x16, 0x17, page 18).

oxullo, 2016-12-02:
@slucero1124 check b2fa5675175d692007cd71f76e3324fb4b2df41b

closed 2016-12-02 by oxullo