oxullo/Arduino-MAX30100 #10: I2C address

tovpe, 2016-11-10:
Great code! :)
I am trying to modify it to work with the MAX30102 sensor. Unfortunately I am a total newbie when it comes to the I2C interface... So I was wondering about 
#define MAX30100_I2C_ADDRESS                    0x57
Where does the 0x57 come from? I'v been looking in the datasheets for the sensor and have found that I2C read and write adress is 0xAE and 0xAF, respectivly, but I'm guessing that is not the same thing..

slucero1124, 2016-11-10:
Hi - 
In my limited experience with I2C, I can tell you that the 0x57 is the address of the actual sensor. You can test it for yourself as well, by using this:

http://playground.arduino.cc/Main/I2cScanner

oxullo, 2016-11-13:
@tovpe, @slucero1124 0x57 is simply a right shift of 0xAE/0xAF. The Wiring/Arduino Wire library expects a 7-bit address for all the methods that address a device (beginTransmission(), requestFrom()). The library takes care of setting (write operation) or un-setting (read operation) the LSB of the address frame.

You find a bit more here: https://www.arduino.cc/en/Reference/Wire
But mostly here: http://www.totalphase.com/support/articles/200349176-7-bit-8-bit-and-10-bit-I2C-Slave-Addressing (8-bit Addresses)

closed 2016-11-13 by oxullo

oxullo, 2016-11-13:
Check also page 25 of the MAX30100 datasheet: https://datasheets.maximintegrated.com/en/ds/MAX30100.pdf
You'll see that their "SLAVE ID" includes also the R/W bit.