oxullo/Arduino-MAX30100 #11: get raw/filtered sensor value

fehlfarbe, 2016-11-16:
Hi,
is there a way to get the pulse and spO2 raw/filtered values? In Debug mode it prints the values to Serial but I want to process it in my main loop.

oxullo, 2016-11-16:
Hey @fehlfarbe 
if you use the PulseOximeter class, there's currently no way to access the unfiltered data.
Such class itself is not really reusable: when I designed it I just wanted to have a higher level component (higher than the sensor abstraction) that could deliver all the consolidated data I required.

If you have to follow a specific implementation pattern, you can use all the ancillary support classes and build your own PulseOximeter, considering that the MAX30100 class exposes raw data. Anyway, since you're not the only one asking for a more scalable component, I'll think about a possible solution.

fehlfarbe, 2016-11-18:
Hey @oxullo
Thanks for your repsonse!
I just added a public member variable to `PulseOximeter` class and update the value in `PulseOximeter::checkSample()`. So I can access the data from main loop. Works good enough for my purpose :)

oxullo, 2016-11-30:
I'm glad you found a way!

closed 2016-11-30 by oxullo