oxullo/Arduino-MAX30100 #12: [Arduino+ MAX30100 (rcwl 0530) ] zeros when extracting SpO2 level and BPM

s1sm1x, 2016-11-29:
Dear Oxullo,

I can easily see that you implemented a really good code, but I'm having a lot of trouble on extracting the SpO2 level and de BPM and I would like you to help me, if you can.

So, I strictly followed the arduino wiring, which is similar to other ones using de I2C interface, no problem here ( I' ve already tried in other breadboards and wires). 
When running the code, the red light on MAX30100 turns on and I'm able to get some values and the loop to go on and on, the problem is that the values are zero for both variables and there is only some different values if I touch the wires or something related. (annex image)

When running on debug mode, I can see some values like 65535 on left and something similar on right to the comma. It seems that the ADC becomes saturated because the left one doesn't even changes. (annex image)

I tried to explain as much as I could and I'm hoping to get the problem resolved soon.

Thank you for the attention and the code :)

s1sm1x

[image]
[image]

oxullo, 2016-12-03:
Hey @s1sm1x do you have anything covering the sensor? (eg: polycarbonate)
Try the new example I've just added here: 177660ff731e4074845a0dc6ad9670e14967a24f (MAX30100_RawData): it allows you to quickly evaluate different current settings for the LEDs. Try to lower both currents till the values are safely out of the maximum values

m-brantheim, 2016-12-18:
I am having a similar problem, although it is not becoming saturated. I just get the same values all the time.

I am not using the MikroElektronika Heart rate click daughterboard, instead using this: [http://www.ebay.com/itm/1PCS-NEW-MAX30100-Heart-Rate-Sensor-Heart-rate-click-/201613701367?hash=item2ef11cf0f7:g:BjIAAOSw-KFXc64t](url)

Could that be a problem since you wrote the code for the MikroElektronika? It is the same MAX30100 sensor in the end so that makes me think it could work.

Connected it to 3,3v and ground and tried to connect the SCL and SDA first to A4,A5 and then the SCL and SDA by the digital pins.

Raw data:
[image]

Debug:
[image]

oxullo, 2016-12-19:
Hey @m-brantheim the values never change? Even when covering the sensor with a finger?
Check the specs of the breakout board you're using, in particular: VIN is 5V or 3.3V? I see two sot-23 units that look like voltage regulators. Check also if the SDA/SCL are properly pulled up.

m-brantheim, 2016-12-19:
@oxullo No I always get the same values, have tried covering the sensor with a finger and on my wrist and hand. I have not been able to find a spec for the breakout board but have contacted customer support so hopefully I can get one. I am connecting the breakoutboard to an Arduino Uno and it has both 5v and 3.3v clearly marked, at first I did not know the sensor should be at 3.3v so I connected it to 5v by accident. I am not sure but this may have damaged the sensor so I need to test with other sensors (have ordered several). By checking that SDA/SCL are properly pulled up do you mean I should do this in the code since Arduino has built in resistors? Like this pinMode(5, INPUT_PULLUP);
I just plugged in the SDA/SCL, and used the example code you wrote. Do you mean I also need to enable resistors in the code or do you mean I have to build this into the circuit?
I am a bit new to electronics so I have a heard time understanding some of this.

oxullo, 2016-12-20:
@m-brantheim the internal pullups are too weak (~20kOhm) for the required specs (plus, I think that they'd be disabled when configuring the pins for TWI operations). Check with a multimeter the resistance between VIN and SDA and VIN and SCL. Just looking at the picture of the board, it looks like that there's a 10k resistor network that might be used also as a pullup.

closed 2017-03-14 by oxullo

sasanfarooji, 2018-02-21:
Hi @oxullo 
I am having problems getting readings from a RCWL-0530 board with a Max30100 sensor.

I have 2 2K ohm resistors connected to SDA and SCL and also another 2K Ohm resister connected to RD, but nothing connect to IRD as it dims the LED significantly when I connect anything to it. once I connected the resistors, I started seeing values on my iphone but other than the battery level the SpO2 and BPM are both zero, it's good to mention that I am using a adafruit feather 32u4 board. 

I have another micro Bus max30100 and when I connect that (without any resistors) it works just fine. 

could you please give me some pointers?

olix86, 2018-08-27:
@s1sm1x I'm having the same issue, did you find a solution?

(sorry to bring this up 2 years later)

olix86, 2018-09-03:
@oxullo You are right, the problem was indeed solved by lowering the LED current (just commenting here in case somebody stumbles upon the same issue)