oxullo/Arduino-MAX30100 #13: Arduino+MAX30100(RCWL-0530) Unable to begin MAX30100

William-An, 2017-01-24:
Hi, thank you for developing such code for MAX30100!

But as my title says, after connecting the sensor to arduino uno, the serial console shows nothing. However, after I unplugged the sensor from my breadboard, the serial console starts to print "Heart rate:0.00bpm / SpO2:0%." After I inserted some "Serial.println();" in  the MAX30100.cpp, I found out that the code stopped at this line: "Wire.write(data);" in "MAX30100::writeRegister" during the "MAX30100::begin" process. 

I really cannot solve, could you help me through this?

William-An, 2017-01-24:
P.S. I cannot detect my MAX30100 through i2CScanner either... After I burnt the i2CScanner into my arduino, the serial console showed nothing, and the code seemed to stop. But after I unplugged the MAX(again), the console started to work

William-An, 2017-01-29:
After checking my MAX30100 datasheet and its SDA & SCL pins' voltage, I realized that it must work under **3.3v of I2C** instead of the **5v on arduino**. I think that is the final reason behind this.

mollyandchloe, 2017-02-10:
I used the rcwl-0530 module, to get it to work I had to cut the 1.8V to the pullup resistors and connect to 3.3V pull up.

Then I had to play with the LED current - Using 4.4 or 7.6 ma Worked well for me.

Thanks for this code, it works great.

oxullo, 2017-03-14:
@William-An did you check this brief schematics? https://github.com/oxullo/Arduino-MAX30100/blob/master/extras/arduino-wiring.pdf

@mollyandchloe thanks for your contribution. Does it really work for you with such low bias current?

ghostmanzero, 2017-03-24:
Hi everyone.i connected RCWL 0530 to arduino. I loaded the code. But it is not working. Rcwl is not working. I think normally red should be operating but as you can see it is not working. And when I open the serial port all I can see is "initializing MAX30100" . can anybody help me? 
[image]

ghostmanzero, 2017-03-24:
Connections :
Vin => arduino 3. 3v 
Scl => ardunio Scl 
Sda=>ardunio sda
Int=> ardunio pin 2 
Gnd=> arduino gnd

ghostmanzero, 2017-03-24:
After connecting the int to the 3.3v (without a pull resistor) now the red led is working. But I still cannot get any result.it doesn't detect the beat. It stops on "pox.begin() ;"  it looks like max30100 is not responding :/.  I have to RCWL. I tried both of them. Nothing changed . I don't know where my mistake is.  I followed the instructions. And I couldn't get a result. I need help

ghostmanzero, 2017-03-25:
@mollyandchloe how did you change the led current?

challee99, 2017-03-27:
use IIC to change RD/IRD current

dokechin, 2017-03-30:
Please teach me.
Is it necessary to connect RD and IRD pins?
Where to connect Arduino pins Number separately?

challee99, 2017-03-30:
OK， I can show you how to connect it . Just.. wait a moment...Get something to handle with..

challee99, 2017-03-30:
I get a meetting...

challee99, 2017-03-30:
[image]

challee99, 2017-03-30:
when IR_LED+/R_LED+ is connected with 3.3V
The IRD/RD should be low or connect to GND. So the RedLED/IRLED can be lighten.

dokechin, 2017-03-30:
Thank,you very much.
After connecting to GND,LED has changed shine stronger.

JAICHANGPARK, 2017-04-01:
I resolved this problem that wasn't operated in arduino UNO. Arduino UNO doesn't have TWI register.
So. Connect 10k ohm register on sdk, scl pin. Then operate Try

dokechin, 2017-04-01:
Thank,you. I tried 1.0KΩ。That going but cannot pick up beat constantly.
 I'will try value up the register.

oxullo, 2017-04-01:
@JAICHANGPARK the Uno equips an ATMega328p which sports a hardware I2C/TWI interface.. And watch out for the pullup, 10k might be too high: with 4,7k I can already see the effects of an RC.
@dokechin check #14 for the beat issue you're experiencing

closed 2017-04-01 by oxullo

sezarosama000, 2017-04-12:
hey can anyone show me the code plz, im thinking maybe mine misses something
@oxullo
@ghostmanzero 
@challee99

sunny950, 2017-04-22:
hi everyone..i have connected max30100 rcw0530 to arduino i loaded the code but in serial monitor its just showing initializing max30100..nothing shows after it can anybody tell me how to fix it.. my pin connections are
Vin => arduino 5v
Scl => arduino analog a5
Sda=>arduino analog a4
Int=> ardunio digital pin 2
Gnd=> arduino gnd

oxullo, 2017-04-22:
@sezarosama000 try the new example MAX30100_Tester on v1.1.0

oxullo, 2017-04-22:
@sunny950 make sure you have proper pullups. Read the README.md and check the proposed schematics in the extras folder

sunny950, 2017-04-23:
@oxullo thank u.i m trying without pullup resistors.should i use two pullup resistor for sda and scl pins seperatly..can u tell me where to connect these pullup ressistor i m new to this project

dokechin, 2017-04-23:
@sunny950
RCW0530 I2C module voltage is 1.8V, so level converter is needed between Arduino and RCW0530. You should buy a voltage level converter board and try again.

sunny950, 2017-04-24:
@dokechin brother in max30100 there is an inside voltage regulator..

dokechin, 2017-04-25:
Please see the topic below, bidirectional level converter is needed.
https://forum.arduino.cc/index.php?topic=447071.0

sunny950, 2017-04-26:
thank u i have tried mega 2560 and its working but its not calcuating sp02 while beat is detected and temperature also showing.what i have to do to calculate spo2.its either have value 0 or 100% which is incorrect i have connected my wires as follow
sda to arduino sda
scl to arduino scl
ird and red common ground with arduino grnd
gnd to arduino ground
[image]

 

[image]

kamran-ali, 2017-07-08:
@sunny950 were u able to solve the problem.
and need some help when i am connecting ird and rd to common ground the red led does'nt blink,
but if i keep ird open the red is blinking. 
and on running test module the red=0 and ir=0.
and do i have to connect interrupt to any digital pin and if it has to be pulled up or down.

kamran-ali, 2017-07-08:
i am using mega2560 so i dont need a pull up register for sda scl, but do i still need it for interrupt ? @oxullo

MAX30100problem, 2017-08-10:
can anyone help me please?
I am trying to get the spo2 using max30100 and arduino nano. I have wired all the pins as an instructions. And the red LED started lighting. But we I am trying to run the code (tester or any of the codes) , i face with this error: Initializing max30100, Failed : I2C error 

to be noted that I wired scl, sdk , int via 4.7k pull up resisitors to vcc (5 volt).

AhmadZMQ, 2017-11-21:
**@challee99** I did it as you said but the problem is both RED and IRLED doesn't light up.when I connect the one with GND (RED led alone) they got light up but  doesn't respond something. What we have to do for.kindly help plz.

sunny950, 2017-11-21:
i have built a sensor for it having ir transmitter and ir reciever ..i
tried om max30100 but cant get any accurate value so i done it through
another method.

liviuanc, 2017-11-21:
The same issue the red and it LEDs are not turning on

liviuanc, 2017-11-21:
I'm using mega with  the same 4.7 k pull up rezistor

challee99, 2017-11-22:
@AhmadZMQ  check your progarm. there is no problem with your LED.

Dwight21, 2017-11-25:
can you give me the wiring for max30100 to arduino mega 2560?

liviuanc, 2017-11-25:
From 5 volt I pulled up the i2c lines with 4.7k rezistors

sunny950, 2017-11-25:
in arduino mega 2560 there are special pins for scl and sda so use that and
for red and ird i have already point the wirings ..

sunny950, 2017-11-25:
use arduino mega 2560 instead ..dont have to use pull up resistors while
using arduino mega 2560

Dwight21, 2017-11-25:
so the  wiring is this VIN->5V, SCL(PULLUP4.7K)->SDL,SDA(PULLUP4.7K)->SDA,IRD&RD->GND,GND->GND. i am right sir ?

Dwight21, 2017-11-26:
can you give me the circuit of it guys . it shows FAILED

Dwight21, 2017-11-26:
@sunny950  when you using your max30100 to arduino your wiring is this VIN->5V, SCL(PULLUP4.7K)->SDL,SDA(PULLUP4.7K)->SDA,IRD&RD->GND,GND->GND.

1bharath1, 2018-02-15:
I too had a problem where it was only printing 0 in all columns.

The fix for it is to just uncomment this line
pox.setIRLedCurrent(MAX30100_LED_CURR_17_4MA);

I came to know about this from

http://edycube.blog.fc2.com/blog-entry-982.html

Shooting77, 2018-02-28:
@challee99 What's IIC ? and how I will use it??

challee99, 2018-03-01:
@Shooting77   Its Communication bus. You should google it .

AhmadZMQ, 2018-03-01:
Hi,
IIC bus is used to make or build a relation or connection between master
and slave to make connection communicable.It is a software based address
transmission.
If you're using one-to-one master slave communication then no need to
define the address of slave else you might be declare the address of all
slave which has to be communicate with master.You should have to use the
SDA and SCL pin to communicate for the same.To get the addess of the slave
you would have to visit their datasheet.

AhmadZMQ, 2018-03-01:
Also you can go through this link https://www.i2c-bus.org/.

Shooting77, 2018-03-01:
@AhmadZMQ thank you, but I connected SDA  & SCL  of MAX30100 to the  SDA(A4)  & SCL(A5) of the Arduino UNO but i still get the (Initializing pulse oximeter..) only without beats.

AhmadZMQ, 2018-03-02:
Hi,
hunmmmmmmmmmmmmmmmm.yeah! I also tried hard to commnicate using the pins as
said by you but got no luck😖.I'm not sure this is the problem of
manufacturing or any other technical issues.I  suggest you a link to
puerchage the new one and get the fun--click o this lin khttps://
learn.sparkfun.com/tutorials/max30105-particle-and-pulse-ox-sensor-hookup-guide

Shooting77, 2018-03-09:
@AhmadZMQ is this (max30105) working with Arduino such as it described in this site https://
learn.sparkfun.com/tutorials/max30105-particle-and-pulse-ox-sensor-hookup-guide

AhmadZMQ, 2018-03-10:
yup....Arduino UNO,Mega, as well as Nano....

Shooting77, 2018-03-10:
Thanks so much @AhmadZMQ

AhmadZMQ, 2018-03-10:
It's my pleasure....have a fun with your electronics👍.

dianjach, 2018-03-26:
I'm having problems. max30100 (rcwl0530) can not work. I'm using your library max30100. and run the max30100 tester and just show it

[image]
[image]

pin max30100 vin = pin arduino v3,3
pin max30100 scl = pin arduino scl ( 4,7 R )
pin max30100 sda = pin arduino sda ( 4,7 R )
pin max30100 int = pin arduino d2 ( 4,7 R )
pin max30100 GND = pin arduino GND

this is like @mesutsaygioglu project.  but I do not understand
can some one help me??

AhmadZMQ, 2018-03-26:
Hi,
Is this working for him( @mesutsaygioglu <https://github.com/mesutsaygioglu>
 ).Also , you can try out by open the INT pin.
As I already told you to just use max30105 (
https://learn.sparkfun.com/tutorials/max30105-particle-and-pulse-ox-sensor-hookup-guide)
breakout board for convenience.I don't know whats wrong with max30100
breakout board,I didn't find any solution for him at yet.

dianjach, 2018-04-09:
i have problem sensor data. the sensor value jumps with a large range. how is the data sensor stable ?? anyone can help

AhmadZMQ, 2018-04-09:
Firstly,just tell me which sensor are you using and let's explain little
bit of you problem....

dianjach, 2018-04-09:
saya menggunakan max30100 rcwl0530.

AhmadZMQ, 2018-04-10:
Hi,
Please purchase a new one MAX301005 from sparkfun.
I  suggest you a link to purchase the new one and get the fun--click o this
link - https://learn.sparkfun.com/tutorials/max30105-particle-
and-pulse-ox-sensor-hookup-guide

Shooting77, 2018-04-14:
Is any one has a library of MPU6050 gyroscope of  for proteus 8??

[20 more replies]