oxullo/Arduino-MAX30100 #14: Missing beats!

DallySon, 2017-01-28:
Dear Sisrs,

Thanks so much for your great job! It is really  helpfull!

But I have problem, my sensor GY-MAX30100 with your library missing my beats. (may be every 2nd beat, or even 2-3 everyting).

I use coverglass (policorb. about 1mm or  lab coverglass 0.1mm - same result).

output looks like:
Beat!
Heart rate:39.55bpm / SpO2:94% / temp:34.31C
Heart rate:39.55bpm / SpO2:94% / temp:34.31C
Beat!
Heart rate:33.36bpm / SpO2:94% / temp:33.00C
Heart rate:33.36bpm / SpO2:94% / temp:33.00C
Beat!
Heart rate:31.18bpm / SpO2:94% / temp:34.00C
Heart rate:31.18bpm / SpO2:94% / temp:34.00C
Beat!
Heart rate:30.39bpm / SpO2:94% / temp:33.56C
Heart rate:30.39bpm / SpO2:94% / temp:33.56C
Beat!
Heart rate:30.17bpm / SpO2:94% / temp:34.56C
Heart rate:30.17bpm / SpO2:94% / temp:34.56C
Beat!
Heart rate:30.10bpm / SpO2:94% / temp:34.31C
Heart rate:30.10bpm / SpO2:94% / temp:34.31C
Beat!
Heart rate:30.07bpm / SpO2:94% / temp:34.81C
Heart rate:30.07bpm / SpO2:94% / temp:34.81C
Beat!
Heart rate:29.95bpm / SpO2:94% / temp:34.56C
Heart rate:29.95bpm / SpO2:94% / temp:34.56C
Beat!
Heart rate:30.01bpm / SpO2:94% / temp:35.13C
Heart rate:30.01bpm / SpO2:94% / temp:35.13C
Beat!
Heart rate:30.03bpm / SpO2:94% / temp:34.50C
Heart rate:30.03bpm / SpO2:94% / temp:34.50C

But real pulse in this time about 80-90bps.

I connect  pin 'INT' of board to my arduino INT0(2).

what can be reason of this problem?

Really hope on your help!

Thanks so much!

DallySon, 2017-01-29:
Maybe I have to correct filter values? ( raw data bigger or smaller than have to be) But I don't know where and how I have to do it. Help me please.

reaper7, 2017-02-04:
I have exactly the same problem with similar values

DallySon, 2017-02-08:
Did you resolve this problem? If so - how?

reaper7, 2017-02-08:
no :( 
we have probably too thin veins ;)

DallySon, 2017-02-08:
may be for thise device :)  but for another thigs - its look good :)
So.. let's awaiting for Wizard  "oxullo" - he will escape us!!  I still hope..

yutakau, 2017-03-07:
Me too. I was wondering just same issue.

oxullo, 2017-03-14:
I have been running tests with at least 12 people. While it runs great for me in any condition, with a very low chance to miss beats, I reckoned that the average performance of the beat detector is definitely low.

Low perfusion (yes, @reaper7) seems to me the primary obvious cause: it causes the S/N ratio to decrease to the point that tiny pressure changes of the finger on the sensor's window cause the beat detector to overshoot, with a long recovery time.

Anybody willing to provide me recordings of the raw data recorded?

DallySon, 2017-03-17:
Thanks for explaning. BUt what  can I  to do - for resolving this issue?

mtx512, 2017-03-18:
I'm using max30100 with a cortex M4, reviewed your algorithm and implemented something similar but heart rates results are inconsistent. The low pass filters seems to work well. Fortunately M4 has fpu so maths operations are faster however I think this may be a problem on a Arduino. The other area is consider is that there can be an buffer overflows on the max30100. Attached are samples from a test run, I'm seeing short periods where IR/RED samples are elongated which causes beat detection to restart, have you experienced this? 1st column in file is a count (ignore), 2nd is RED and 3rd is IR.
[output_max30100_2.txt](https://github.com/oxullo/Arduino-MAX30100/files/852589/output_max30100_2.txt)

oxullo, 2017-03-26:
@mtx512 thanks for the data and the hint, I'll be looking at them next week. Does the data come from your port? Can you link the fork? And did you scope-checked if your code runs in less than 10ms?

oxullo, 2017-03-26:
@DallySon I'll be adding a public interface for setting the bias current of the IR LED, so you can experiment easily with different values. If you can't wait (sorry but I have very limited amount of time), just modify the library, changing the constant IR_LED_CURRENT to something less than the current 50mA

https://github.com/oxullo/Arduino-MAX30100/blob/master/src/MAX30100_PulseOximeter.h#L24

DallySon, 2017-03-27:
Thanks so much. I have time now to wait! Just don't forget about me.

oxullo, 2017-04-01:
@DallySon you'll find the newly added interface on v1.0.1

oxullo, 2017-04-01:
@mtx512 I timed the execution of the time-critical code and it stays always below 500µs (ATMega328p @16Mhz). There's plenty of time for 100Hz sampling, even though I noticed a fairly irregular jitter on the wait code. I'll try to see whether I can move to a timer. If you want to elaborate on another ticket, feel free to open it!

closed 2017-04-01 by oxullo

pradeeps15, 2018-02-06:
Firstly, a great library and the forum has been really helpful.
I am having problem with this sensor and getting incosistent heartbeat. Mostly getting stable SPO2 at 93% to 96% though. I was wondering if adjusting the current might help. Also, I must add that I have gone through a lot of issues raised earlier to troubleshoot my RCWL 0530 board to make it work on the Arduino UNO R3 and it does pass the Tester code and present with some readings in the Minimal example and another code based on your libraries by [14core](http://www.14core.com/wiring-the-max30100-heart-rate-monitor-with-arduino-microcontroller/). Is it necessary to cover the MAX30100 IC with a glass cover to get consistent results? I will debugging using the Rolling_Graph. I am new to Arduino and MAX30100 and could really use the help. Thanks.

FarzanehK, 2020-07-27:
> Mostly getting stable SpO2 at 93% to 96% though.

Dear Pradeeps15
We have the same problem with green MAX30100, even for hypoxia patient, the SpO2 does not decrease to less than 93%. Would you please let me know if you have solved your problem about SpO2 values?

Bests