oxullo/Arduino-MAX30100 #16: Sensor doesn't work properly :/

ghostmanzero, 2017-03-25:
First of all, Unfortunately it didn't work with 3.3v.  I connected  it to 5v and used 10k pull ups for scl and sda. And Int connected to 3.3v directly (without a pull up)(it doesn't matter whether it is connected to 3.3v or not. I tried both of them) it doesn't detect beats when I put my findger on sensor. It shows "Heart rate:0.00bpm / Spo2 :0%" but when I pull my finger back for 1 or 2 cms above, it detect the beats. But values are not right. Random things.like  63, 100,151,... 
İ  have to RCWL and tried both them too. But it is still the same :/

oxullo, 2017-03-26:
When you say that it didn't work with 3.3V, which pullups for SDA/SCL did you use?
I guess that if you used 10k it might be too large for the 400kHz data rate, depending on how you wired up the board. Use 4,7k or less, the datasheet limits only the lower bound which is 500Ohm (NOTE: only for the I2C interface, not for the interrupt line).
And another guess is that it starts reporting when you change the capacitance of the package by pulling the finger away..

Follow the guidelines you see here: https://github.com/oxullo/Arduino-MAX30100/blob/master/extras/arduino-wiring.pdf

ghostmanzero, 2017-03-26:
Thank you so much :)  I found the problem. I think brightness  of red led didn't let the Ir to get the beats. I reduced the led current. Now it is working.  Values are not great but it is working anyway(do you have a suggestion to get better results?)  . I think it is because of the senior's characteristic. And the code is working very well :) thank you for the this great code and replying :)

oxullo, 2017-03-26:
Glad that you have it working. I'll be adding a public interface for adjusting the bias current of the IR LED so no modifications of the library will be needed

closed 2017-03-26 by oxullo

saikapilan, 2018-03-01:
I am using RCWL max30100 I have problem in glow led. If I connect IR and irD with ground it start glow but in program it just initializing max30100

saikapilan, 2018-03-01:
The connection are 
Vin to 5v
Sda to A4(with resistor 4.7)
Scl to A5(with resistor 4.7)
Int no connection
IR to ground ( with resistor)
IRD to ground( with resistor)
Gnd to gnd

HemavathiEaswaran, 2019-02-16:
i have a problem with max30100 sensor if i keep my finger on the sensor it shows some values and if i keep my mobile or some object  it  shows some values .i detects every thing and gives values.i didnot connect resistors ,capacitors externally. what can i do to solve this ?

celciusjj, 2019-05-07:
> The connection are
> Vin to 5v
> Sda to A4(with resistor 4.7)
> Scl to A5(with resistor 4.7)
> Int no connection
> IR to ground ( with resistor)
> IRD to ground( with resistor)
> Gnd to gnd

Me too, 
What solution did you give him?

ElectronicsInnovation, 2021-03-18:
This video might help you, in this video, an Indian guy explained the issue in detail with the help of the max30100 sensor schematic diagram, and solved it in two ways. https://youtu.be/ZqdmA4NAqb0