oxullo/Arduino-MAX30100 #18: Is there any way to enter sleep mode?

viggi1000, 2017-04-14:
Hi, Thanks for your amazing library man!

I used it for a [low cost neonatal incubator](https://github.com/viggi1000/ADITI_Blynk101Hackster). I was actually having issues with entering low power modes. I want to take readings every 2 minutes and then go to sleep but the MAX30100 led is still blinking, I did try changing the LED current function to 0 but that screws up the HR and SP readings. I was wondering if there is any function to exit pox.begin() and turn off the LED and then reset the BPM and SP readings?

Sorry if I am being a bit too vague or silly, I'm rather new to firmware dev.

oxullo, 2017-04-22:
Hey @viggi1000 thanks for using the library! I added two methods: shutdown/resume (to both base sensor class and mirrored upstream on the PulseOximeter helper).
As stated on page 15 of the MAX30100 datasheet (https://datasheets.maximintegrated.com/en/ds/MAX30100.pdf), putting the unit into shutdown mode preserves the registers states. So you have just to invoke shutdown() when you want to put the sensor to sleep and resume() when you're ready to read new samples. Could you test it, please?
You'll find them on commit 903ec2

oxullo, 2017-04-22:
I've just tagged a new version, so you'll find it on v1.1.0

viggi1000, 2017-04-22:
I will check it out at the soonest and let you know. Can't thank you enough
actually!

oxullo, 2017-05-11:
@viggi1000 I'm closing the issue, feel free to reopen if needed

closed 2017-05-11 by oxullo