oxullo/Arduino-MAX30100 #19: Wrong data

billytoft0, 2017-04-20:
Hi, 

I'm using the MAX30100_Minimal code and the data I'm receiving is all wrong. As soon as I connect the circuit, even without touching the sensor, it starts to print SpO2 and heart rate values and it also gives an extremely high temperature value.

Is there anything you know I could modify to rectify this?

oxullo, 2017-04-22:
Hey @billytoft0 check v1.1.0, I added a tester (the example MAX30100_Tester) that should help to track the issue. Please run it and eventually post the output here.
Make sure you also check the troubleshooting notes in the readme.

krisscaroli, 2017-05-09:
hi! help me. I get error ,IDE ARDUINO say: " [FAILED"](url
[image]
)

oxullo, 2017-05-11:
@krisscaroli did you check the readme, the schematics suggestion and double-checked your i2c configuration? If so, which sensor breakout are you using and how did you connect it to the arduino and which arduino?

krisscaroli, 2017-05-20:
Logre solucionarlo Oxullo pero ahora los valores de Sa02 salen 94% hasta 95% cuando en un oximetro real salen 97 y 98% a que se debe?

uutzinger, 2017-05-29:
I believe the code to calculate SpO2 has an issue as it might be based on a ratio of the log(MAX-MIN) instead of log(MAX/MIN). I would be happy to share class notes I made and to compare and learn more about the algorithm used in this excellently written driver. utzinger@email.arizona.edu.

krisscaroli, 2017-06-06:
hi uutzinger, I  wrote an emai, help mel!!

oxullo, 2017-06-17:
SpO2 calculation is basically a copy of the algo I found on the referred TI appnote.
It's quite sensitive to the current bias of the red LED, I had no proper mean to conduct a reliable test set in order to improve it.

@uutzinger the algo works by calculating the ratio of the RMS signals of red and IR spectra. If you're still willingly to share your findings, feel free to open a ticket and describe your approach, so also other people have the chance to intervene. Thanks in advance.

Here you can find a detailed analysis of the approach: https://morf.lv/implementing-pulse-oximeter-using-max30100
And this is where I snatched the algo: http://www.ti.com/lit/an/slaa274b/slaa274b.pdf

I'm going to close this issue since the topic is drifting from the problem of the OP.

closed 2017-06-17 by oxullo

jatinaroraece, 2017-08-28:
Hello Oxullo, I am using Minimal sketch and getting the required readings. However, the readings are always nearly 30BPM (heartbeat) which is not correct. How to solve this problem. I think scaling won't work. Moreover, Spo2 value is also 94% constantly.  Thank You

selcukartut, 2017-09-08:
hi jatinaroraece, I am having the same readings around 30BPM, have you been able to find a solution for that? Thanks in advance

jatinaroraece, 2017-09-08:
@selcukartut I have not solved it yet. I am waiting for the response of @oxullo to find the optimum solution.

oxullo, 2017-09-17:
@jatinaroraece @selcukartut is the sensor covered by any material, whatever it is?
Tweak the library by changing this value: IR_LED_CURRENT (MAX30100_PulseOximeter.h:24). Try halving the value (eg: MAX30100_LED_CURR_24MA) and check if it gets better. Try also other values in order to improve the reliability of the readout (eg: bisect between min and 24mA and between 24mA and 50mA). Please let me know the results!

oxullo, 2017-09-18:
Pardon, I was looking at an old version. Make sure you have 1.1.0 and you can use the method PulseOximeter::setIRLedCurrent() to set the bias current of the IR LED instead of tweaking the library.