oxullo/Arduino-MAX30100 #21: IoT Confusion

brlisows, 2017-05-01:
I am trying to use the MAX30100_Minimal code and add to it to connect to ThingSpeak, an IoT platform. I added the same lines of code that I have used for previous i2c sensors to send data to Thingspeak, but for some reason when I add them to the MAX30100 code the serial monitor begins displaying values of 0 for beats for minute and SpO2 percentage. The temperature continues to read correctly, so I do not know what the problem is?

oxullo, 2017-05-11:
@brlisows I imply that when you don't add the code it works properly?
Did you try MAX30100_Tester example?

Anggato, 2017-05-20:
yes @brlisows  i have same problem maybe someone help me

Anggato, 2017-05-21:
@oxullo could yo help me ?

oxullo, 2017-06-17:
@Anggato I need more info. Did you try the tester?

oxullo, 2017-08-05:
Closing due to inactivity. Feel free to open a new one

closed 2017-08-05 by oxullo

shashank051996, 2018-03-11:
@oxullo Same problem MAX30100 is working fine when used without involving any IoT i.e. if we don't use the thingspeak part of code but involvement of thingspeak gives only 0 as its readings. Please resolve

shashank051996, 2018-03-11:
Its very urgent. Please resolve it fast

12awan, 2018-03-12:
I tried to use the code MAX30100_Minimal and added it to connect to ThingSpeak, an IoT platform. I added the same line of code that I used for the previous i2c sensor to send data to Thingspeak, but somehow when I added it to the code MAX30100, the serial monitor started displaying the value 0 for the tick for percentage of SpO2 not reading, but when use MAX30100_Tester why not can upload in arduino and nodemcu, please help

izabellelima, 2018-04-15:
hello, some news about integration with thingspeak?

Marinamyat24, 2018-07-02:
Hello, is there any news about this problem. And I also cannot upload MAX30100_Tester. I'm also facing this problem and this project is really urgent for me. Please help me. Thank you in advance.

pabloesya, 2018-12-30:
Hello , any solution for that problem ?