oxullo/Arduino-MAX30100 #22: I2c error in tester

nghiatruong1204, 2017-05-11:
@oxullo  this library is really heplful, tks bro :)))
can u teach me how to fix I2c error found by MAX30100_tester pls  :))

oxullo, 2017-05-11:
@nghiatruong1204 which error? :) Post it here and please add all the details you can, including: which breakout and arduino you're using, which arduino framework/version, how did you connect the breakout.
Check thoroughly the readme, it holds some relevant info to perform an initial self check. Usually the main issue is the lack of proper pullups on the SDA/SCL lines.

nghiatruong1204, 2017-05-11:
@oxullo 
here is my arduino nano and max30100, i use arduino arduino IDE 1.8.1

nghiatruong1204, 2017-05-11:
@oxullo 
this is how i connect my breakout 
VIN connected to 1.8v (converted from 3.3v )
SCL to A5 and SDA to A4 (yellow and orange wires)
INT to D2 
SCL SDA and INT are equipped with pullups (4.7k ohm resistors)
IRD to 3.3v and RD to GND (3.3v power is provided by arduino nano)
GND to common GND by arduino nano
I will give you more in detail by pictures of my breakout :)))

oxullo, 2017-05-11:
@nghiatruong1204 to which rail the pullups are connected to?

nghiatruong1204, 2017-05-11:
@oxullo hope this gonna be helpful xDDD  
[image]

nghiatruong1204, 2017-05-12:
i think there is no problem with the sensor, only I2C error :((
[image]

nghiatruong1204, 2017-05-12:
I2C error solved but _**heart rate**_ and **_SP02_** go wrong :) 
do u have any solution for this ? i need this done for school project xDDD

[image]
[image]

oxullo, 2017-05-12:
How did you solve the primary issue? please leave it described it here for the community!
Regarding the readout, check the new hints I added to https://github.com/oxullo/Arduino-MAX30100/blob/master/README.md (last paragraph).

nghiatruong1204, 2017-05-12:
i changed all the wires and different breadboard bro :)
then i used the power provided by arduino nano for VIN, result in the brighter IR led 
still working on readouts xDDD

oxullo, 2017-05-12:
Glad that you found out. Good luck with experimenting the bias current, that might be tricky!

closed 2017-05-12 by oxullo

nghiatruong1204, 2017-05-14:
[tester IR and Rd result.txt](https://github.com/oxullo/Arduino-MAX30100/files/999742/tester.IR.and.Rd.result.txt)
what can these data tell about the sensor?

x3bachi, 2017-05-30:
Same problem here, I get 0 reading for heart rate and SpO2.
please help

nghiatruong1204, 2017-05-30:
@x3bachi  since i unplugged IRD it works perfectly 
u just need to wire from RD to GND and follow wiring instruction form this library :))))

x3bachi, 2017-05-30:
Thank you @nghiatruong1204 you replied fast :)
it works now but the reading is weird some times reads high values and sometimes reads low values also it skips some beats

x3bachi, 2017-05-30:
Also the sensor keeps​ getting hot to the point i can't touch it

AsmaaSayed, 2021-01-30:
Hi @x3bachi
I have the same with MAX30100 that it's reads are non logical 
Can you help me if you solved it in the past

x3bachi, 2021-01-30:
Hi Asmaa,
It's been a long time since I worked with this sensor, but I
remember fixing it.
the schematics that came with it were wrong if you follow it some of the
pins will be short-circuited.
If you have many sensors try playing with changing the pins and hopefully
none of them will burn xD.
Good luck with your project :).

AsmaaSayed, 2021-01-30:
Ok
Thanks very much for your help