oxullo/Arduino-MAX30100 #23: Failed Initialization Test: Wrong Part ID

tiahunt, 2017-06-19:
### Troubleshooting checklist

- [x] I read the README (on master) thoroughly
- [x] I ran the MAX30100_Tester and I'm going to paste the output down below
- [x] I filled in all the details of my setup down below

### Description of the issue
Ran MAX30100_Tester and received the following.

### Output from MAX30100_Tester example
[image]

### Details of my setup
Setup follows schematic. SDA in A4 and SCL in A5. INT was not used.

* Arduino hardware: Arduino Nano
* MAX30100 breakout: ---
* Arduino framework version: Arduino 1.8.2
* MAX30100 library version: v1.1.0

oxullo, 2017-06-19:
@tiahunt thanks for the detailed issue report. The reason is quite simple: you're using a MAX30102 and not a MAX30100. The former is, unfortunately, not supported by the current driver. Please confirm

tiahunt, 2017-06-19:
That seems to be the issue as my sensor worked fine using MAX30102 code. I must have some sort of variant that was incorrectly labeled. Thank you for your help!

oxullo, 2017-06-20:
You're welcome!

closed 2017-06-20 by oxullo

AhmadZMQ, 2018-07-20:
@**tiahunt** from where you get the code for MAX30102,because I'm encountered with same issue.Here the one thing is ,name on the board is MAX30100 and gives  "**_Initializing MAX30100..FAILED: wrong part ID 0x15 (expected: 0x11_)** " error....how can I resolve it??
**@oxullo**  you can. also .

Islamaim, 2019-11-20:
Hi, After some tries, It works with this library https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library

gitakash04, 2020-09-18:
@Islamaim  thanks a lot dear.....it worked. Love 0000. stay blessed

MKowalski234, 2023-02-20:
Hello, i'm trying to use my DFRobot MAX30102 and it's not being found by builtin code from said library listed above, i ran I2C scanner and it's showing 0x57 which means part is fine, if you guys are still out there could you help me out ?