oxullo/Arduino-MAX30100 #24: Wiring for NodeMCU-esp8266

anwarbashir, 2017-07-31:
### Troubleshooting checklist

- [ ] I read the README (on master) thoroughly
- [ ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ ] I filled in all the details of my setup down below

### Description of the issue

### Output from MAX30100_Tester example

### Details of my setup

* Arduino hardware: nodemcu-esp8266
* MAX30100 breakout: MAX30102/MAX30100 Heart Rate Breakout Sensor Blood Oxygen Transducer for Arduino
* Arduino framework version: 1.8.3
* MAX30100 library version: 1.1.0
I have attached 3.3V and GND. I have connected SDA - GPI04(D2) nd SCL - GPI05(D1). I have added the line Wire.begin(4, 5); //sda, scl. I ran the MAX30100_Tester code. I am getting FAILED: I2C error. Do you think I need pull-up resistors on SDA/SCL or am I using wrong pins or is something else wrong?

Would be graet if I could get your code working with my nodemcu-esp8266.

oxullo, 2017-08-05:
Hey @anwarbashir in the readme it is explicitly stated that the most critical thing to check are the two 4.7kOhm pullups on SDA/SCL lines..
Note also that MAX30102 is not supported, so make sure you have a MAX30100 on your breakout.
I do not support esp8266, but if you make it working please let me know

anwarbashir, 2017-08-07:
Hi, sincere apologies, I should have read the information properly. I am
trying to get an ESP32/ESP8266 to interface with MAX301002 that has
breakout with circuitry already attached. To begin the process of
developing a heart-rate monitor interface I have connected to a Raspberry
Pi and using an IC2 library available at
http://wiringpi.com/reference/i2c-library/ I am begining the journey. I
have published my code on github at
https://github.com/anwarbashir/max301002-raspi. At the moment it simply
obtains the part id (I think it works but I am checking). Please feel free
to keep an eye on this... Again my apologies for not reading your
documentation properly even though you have made the process clear.

oxullo, 2017-08-12:
No prob! I'll keep an eye on your repo!

closed 2017-08-12 by oxullo

madhujiot, 2017-12-11:
hi, I guess am able to interface with MAX30100 module using NodeMCU ESP12 config. Am getting HeartRate and SPO2 value(%) appropriately as expected. Please let me know for any help and if you are still stuck with this. Will post the details soon.

Praveen-dhanasekae, 2018-01-09:
@madhujiot can you please help me with those codes ? i m stuck withit , i dont find libraries to use.

madhujiot, 2018-01-09:
Sure, I am stuck with some other work. In couple of days, I will post there.

Praveen-dhanasekae, 2018-01-09:
thank you @madhujiot

Praveen-dhanasekae, 2018-01-12:
@madhujiot i really need this . Can you upload it faster? i need it to move to the next step .My schedule to finish it is very less .So please make it faster.

madhujiot, 2018-01-14:
really Sorry as I wanted to send now, but before that I thought I will
check with my Nodemcu, its crashing due to some library issues or some
other recent updates on my IDE, not sure.. will rectify and send.

Praveen-dhanasekae, 2018-01-15:
ok @madhujiot try to send it soon ... can you tell me like what were the libraries used by you ?i ll try it from my part..

madhujiot, 2018-01-15:
hi Praveen,

I used MAX300 library from OXullo only (MAX30100_PulseOximeter.h) with
PulseOximeter as the interface and wire.h for I2C operations.

I need to cleanup a lot my laptop before I re-instantiate the Arduino and
sensors setup, it may take a while.,

karimiahmad, 2018-01-22:
Hi @madhujiot  ,@oxullo , @anwarbashir 

I want to run MAX30100 with nodeMCU. Can you upload the updated code?
https://www.sunrom.com/m/5337
https://www.banggood.com/V3-NodeMcu-Lua-WIFI-Development-Board-p-992733.html?cur_warehouse=CN

Thanks a lot.

esz11, 2018-02-23:
Hi @madhujiot  
1.Did you add pull up resistors to the SDA and SCL lines ? 
2.Did you make any modifications to the MAX30100 library ? 

Thanks

Phi-Van-Lam, 2018-07-15:
Thanks!
I tried but not success!
Could you help me by how to run this programe on my Pi?

Phi-Van-Lam, 2018-07-15:
Not success on my Pi.
Could you send me your code?