oxullo/Arduino-MAX30100 #26: Issue with Wemos D1 mini and MAX30100

jonportus, 2017-08-25:
I've tested your library with a Wemos D1 and a MAX30100 breakout board and I'm getting the following error any ideas why ? I'm using the 3.3v from the D1 and all others connections as you would expect except I'm not using the INT pin.

Jonathan

Failed to initialize the HRM sensor
ERROR: Failed to initialize pulse oximeter

Soft WDT reset

ctx: cont 
sp: 3ffef3a0 end: 3ffef580 offset: 01b0

>>>stack>>>
3ffef550:  3fffdad0 3ffee530 3ffee38c 4020210c  
3ffef560:  feefeffe 00000000 3ffee54c 40203810  
3ffef570:  feefeffe feefeffe 3ffee560 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld

nurfaizfoat, 2017-09-18:
@jonportus try to use that very MAX30100 with other MCU in your keeping. And please share how's your wiring/pinout with this setup.

FYI, I'm running the same setup without no problem at all

oxullo, 2017-11-19:
Closing this issue for inactivity and because I do not support ESP mcus.

closed 2017-11-19 by oxullo