oxullo/Arduino-MAX30100 #27: Stuck at "MAX30100 Initializing.."

nazmibojan, 2017-09-18:
### Troubleshooting checklist

- [OK] I read the README (on master) thoroughly
- [OK] I ran the MAX30100_Tester and I'm going to paste the output down below
- [OK] I filled in all the details of my setup down below

### Description of the issue
Hello,
I just run MAX30100_Tester and get stuck in serial monitor at "Initializing MAX30100.."
Why was that happened?
My MAX30100 - Arduino Wiring:
VIN -> 3.3V
GND -> GND
SDA -> SDA
SCL -> SCL
INT -> D2

Regards,
Nazmi
[image]

### Output from MAX30100_Tester example
Initializing MAX30100..

### Details of my setup

* Arduino hardware: Arduino Uno R3
* MAX30100 breakout: https://www.tokopedia.com/caravane/max30100-pulse-oximeter-heart-rate-sensor-module
* Arduino framework version: Arduino IDE 1.8.4
* MAX30100 library version: v1.1.0

oxullo, 2017-09-18:
This is most likely to be an I2C communication issue.. As far as I can see from the photo, the module you're using equips already the pullups resistors for the I2C bus, but check if you have 3V3 on SDA and SCL lines with no running firmware.

oxullo, 2017-09-18:
Check also #22 
In theory, using 3.3V for the I2C communication instead of a 5V on a 5V device, might introduce such issues. MAX30100 datasheet indicates 5V as maximum VIN for the two LEDs, so you could try to switch VIN of the module to the Arduino's 5V. But make sure the voltage regulator unit of your breakout board is rated for it.

nazmibojan, 2017-09-18:
Thanks @oxullo !
You're right that this is an issue of voltage at SDA and SCL.
I used Logic Level Converter to solve this problem.
This is an example of wiring Arduino and my MAX30100 Module:

[image]

Regards,
Nazmi

oxullo, 2017-09-18:
Thanks for posting your solution! I'll add a note to the troubleshooting

closed 2017-09-18 by oxullo

fawadrashid, 2020-09-27:
> Thanks @oxullo !
> You're right that this is an issue of voltage at SDA and SCL.
> I used Logic Level Converter to solve this problem.
> This is an example of wiring Arduino and my MAX30100 Module:
> 
> [image]
> 
> Regards,
> Nazmi

Can you provide code of your sketch as well ?

fawadrashid, 2020-10-01:
> > Thanks @oxullo !
> > You're right that this is an issue of voltage at SDA and SCL.
> > I used Logic Level Converter to solve this problem.
> > This is an example of wiring Arduino and my MAX30100 Module:
> > [image]
> > Regards,
> > Nazmi
> 
> Can you provide code of your sketch as well ?

Never mind I found the code. Thanks for the tip on using a logic converter

mayur-003, 2024-02-07:
Can you provide code for it

mayur-003, 2024-02-07:
> Never mind I found the code. Thanks for the tip on using a logic converter
Give the code please