oxullo/Arduino-MAX30100 #29: Wrong values

adepalo, 2017-10-20:
Hi
I'm using a GY-MAX3010 and I've noticed wrong values `�BInitializing pulse oximeter..SUCCESS
Heart rate:0.00bpm / SpO2:0% / temp:33.13C
Heart rate:0.00bpm / SpO2:0% / temp:34.00C
Heart rate:0.00bpm / SpO2:0% / temp:34.00C
Beat!
Heart rate:29.52bpm / SpO2:0% / temp:32.63C
Beat!
Heart rate:48.40bpm / SpO2:0% / temp:32.63C
Heart rate:48.40bpm / SpO2:0% / temp:32.63C
Beat!
Heart rate:35.33bpm / SpO2:94% / temp:32.63C
Heart rate:35.33bpm / SpO2:94% / temp:32.69C
Beat!
Heart rate:31.88bpm / SpO2:94% / temp:32.69C
Heart rate:31.88bpm / SpO2:94% / temp:32.69C
Beat!
Heart rate:30.79bpm / SpO2:94% / temp:32.69C
Heart rate:30.79bpm / SpO2:94% / temp:33.69C
Beat!
Beat!
Heart rate:51.09bpm / SpO2:94% / temp:33.69C
Beat!
Heart rate:62.31bpm / SpO2:94% / temp:33.31C
Beat!
Heart rate:84.61bpm / SpO2:94% / temp:33.31C
Heart rate:84.61bpm / SpO2:94% / temp:33.19C
Beat!
Heart rate:40.21bpm / SpO2:94% / temp:33.19C
Heart rate:40.21bpm / SpO2:94% / temp:32.88C
Beat!
Heart rate:33.35bpm / SpO2:94% / temp:32.88C
Heart rate:33.35bpm / SpO2:94% / temp:32.50C
Beat!
Heart rate:31.22bpm / SpO2:94% / temp:32.50C
Beat!
Heart rate:31.49bpm / SpO2:94% / temp:33.31C
Heart rate:31.49bpm / SpO2:94% / temp:33.31C
Beat!
Heart rate:35.66bpm / SpO2:94% / temp:32.81C
Beat!
Heart rate:58.77bpm / SpO2:95% / temp:32.81C
Beat!
Heart rate:43.20bpm / SpO2:95% / temp:33.50C
Beat!
Heart rate:76.64bpm / SpO2:95% / temp:33.50C
Heart rate:76.64bpm / SpO2:95% / temp:33.31C
Beat!
Heart rate:39.73bpm / SpO2:94% / temp:33.31C
Heart rate:39.73bpm / SpO2:94% / temp:33.81C
Beat!
Heart rate:33.21bpm / SpO2:94% / temp:33.81C
Heart rate:33.21bpm / SpO2:94% / temp:33.31C
Beat!
Heart rate:31.85bpm / SpO2:94% / temp:33.31C
Beat!
Heart rate:37.49bpm / SpO2:95% / temp:34.38C
Beat!
Heart rate:57.87bpm / SpO2:95% / temp:34.38C
Heart rate:57.87bpm / SpO2:95% / temp:33.38C
Beat!
Heart rate:37.09bpm / SpO2:95% / temp:33.38C
Beat!
Heart rate:34.44bpm / SpO2:94% / temp:33.81C
Heart rate:34.44bpm / SpO2:94% / temp:33.81C
`

Run the test 

`Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=31.62C
All test pass. Press any key to go into sampling loop mode
`
Then run the minimal again

`��c
�Initializing pulse oximeter..SUCCESS
Heart rate:0.00bpm / SpO2:0% / temp:31.87C
Heart rate:0.00bpm / SpO2:0% / temp:33.38C
Beat!
Heart rate:34.65bpm / SpO2:0% / temp:33.38C
Heart rate:34.65bpm / SpO2:0% / temp:32.13C
Beat!
Beat!
Heart rate:54.65bpm / SpO2:94% / temp:32.13C
Beat!
Heart rate:68.31bpm / SpO2:94% / temp:32.81C
Beat!
Heart rate:103.46bpm / SpO2:94% / temp:32.81C
Heart rate:103.46bpm / SpO2:94% / temp:33.00C
Beat!
Heart rate:41.83bpm / SpO2:94% / temp:33.00C
Heart rate:41.83bpm / SpO2:94% / temp:33.63C
Beat!
Heart rate:33.78bpm / SpO2:94% / temp:33.63C
Beat!
Heart rate:33.48bpm / SpO2:94% / temp:33.63C
Heart rate:33.48bpm / SpO2:94% / temp:33.63C
Beat!
Heart rate:37.77bpm / SpO2:95% / temp:32.38C
Beat!
Beat!
Heart rate:63.69bpm / SpO2:95% / temp:32.38C
Heart rate:63.69bpm / SpO2:95% / temp:32.38C
Beat!
Heart rate:56.15bpm / SpO2:94% / temp:32.38C
Beat!
Heart rate:39.60bpm / SpO2:94% / temp:32.81C
Heart rate:39.60bpm / SpO2:94% / temp:32.81C
Beat!
Heart rate:37.54bpm / SpO2:94% / temp:32.88C
Beat!
Heart rate:62.24bpm / SpO2:95% / temp:32.88C
Beat!
Heart rate:44.12bpm / SpO2:95% / temp:33.50C
Beat!
Heart rate:77.72bpm / SpO2:95% / temp:33.50C
Heart rate:77.72bpm / SpO2:95% / temp:32.69C
Beat!
Heart rate:39.69bpm / SpO2:95% / temp:32.69C
Heart rate:39.69bpm / SpO2:95% / temp:33.00C
Beat!
Heart rate:33.20bpm / SpO2:95% / temp:33.00C
Heart rate:33.20bpm / SpO2:95% / temp:32.50C
Beat!
Heart rate:31.16bpm / SpO2:95% / temp:32.50C
Heart rate:31.16bpm / SpO2:95% / temp:32.69C
Beat!
Heart rate:30.41bpm / SpO2:95% / temp:32.69C
Heart rate:30.41bpm / SpO2:95% / temp:32.69C
Beat!
Heart rate:30.13bpm / SpO2:95% / temp:32.69C
Heart rate:30.13bpm / SpO2:95% / temp:33.31C
Beat!
Heart rate:30.01bpm / SpO2:95% / temp:33.31C
Heart rate:30.01bpm / SpO2:95% / temp:32.38C
Beat!
Heart rate:29.88bpm / SpO2:95% / temp:32.38C
Heart rate:29.88bpm / SpO2:95% / temp:32.19C
Beat!
Heart rate:29.91bpm / SpO2:95% / temp:32.19C
Heart rate:29.91bpm / SpO2:95% / temp:32.69C
Beat!
Heart rate:29.93bpm / SpO2:95% / temp:32.69C
Beat!
Heart rate:35.71bpm / SpO2:94% / temp:32.69C
Heart rate:35.71bpm / SpO2:94% / temp:32.69C
`

[image]
[image]

Tried to change 
  // pox.setIRLedCurrent(MAX30100_LED_CURR_7_6MA);
but I had no realiable data

Using Arduino Mega 2650

Wiring:
4.7K pull-up on SDA and SCL
4.7K pull -up on INT -> Pin 2
VIN 3.3V Pin 
All pull ups going to 3.3V

Can you please help me?

In case I'll have to change the sensor, can you please suggest me where to buy a good one?

njordan77, 2017-11-07:
I have the same issues with 3 boards....bpm is showing values arround 30bpm whcih is far from realistic. also tried 50mA and 7,6mA. Any tipps what i could try?

novel-notion, 2017-11-13:
I am having the exact same issue with an esp8266. I tried the Sparkfun Library that says its for the "MAX30x" implying that it can work on the 301 but no dice. The LED doesn't even light up.

bobybobybob, 2017-11-15:
I have the green Board one from China. It has labe  RCWL-0530  backside and front MAX30100. There are a smd 65K5 and N1IF 

I use CHINA UNO Clone 5V

RED LED not work with 3.3V or 5V. I2C Scanner found nothing

RED Led work with 5V but also need one 4,7kO resistor from VCC to SDA and another from VCC to SCL and I2C Scanner say OK address: 87 (0x57)

Some Demo Programms do work. After power on, the RED Led need some seconds to turn on. Without connected i2C it wont turn on. 

 When all is OK it show wrong values. BPM most about 30-35,  SpO2 about 93-95 maybee ok

With 10kO from VCC to SDA and another from VCC to SCL. 

Testcode MAX30100_Tester  write

Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=24.50C
All test pass. Press any key to go into sampling loop mode

**Is the Sensor-Item rubbish or the libary have bug?**

oxullo, 2017-11-19:
@adepalo @bobybobybob @njordan77 please have a look at the new notes I added to the main readme: https://github.com/oxullo/Arduino-MAX30100/blob/master/README.md
In particular, make sure you check the two extras recoder and rolling_graph
If none of the tools helps you finding a way to adjust the behaviour of the sensor, please post here a recorded session so I can run an analysis. More info on how to record a session here: https://github.com/oxullo/Arduino-MAX30100/tree/master/extras/recorder

oxullo, 2017-11-19:
@adepalo I always used this breakout: https://shop.mikroe.com/heart-rate-click
I'm not endorsing Mikroelektronika and I haven't had the chance to test any other breakout.

oxullo, 2017-11-19:
@adepalo please check if the branch https://github.com/oxullo/Arduino-MAX30100/tree/issues/%2332/workaround solves your issue

bobybobybob, 2017-11-20:
Rolling Graph say Arduino IDE
    maxv = max(maxv, abs(max(samples)), abs(min(samples)));
exit status 1
macro "max" passed 3 arguments, but takes just 2

Here is Data from the RAW Programm... Copy from Arduino Serial Monitor.
Nofinger.txt   no Finger put on sensor
3 Time Finger.txt   3 different finger has placed on sensor

[nofingers.txt](https://github.com/oxullo/Arduino-MAX30100/files/1489452/nofingers.txt)
[3 time finger.txt](https://github.com/oxullo/Arduino-MAX30100/files/1489453/3.time.finger.txt)

njordan77, 2017-11-21:
tried the issue test version, still same incorrect values on 3 boards...arround 30bpm but far from realistic values arround 60-80. sometimes you see also values higher than 40 but most of the time not. Temp / SPO2 look OK. Thanks

oxullo, 2017-11-21:
@bobybobybob the rolling graph must be opened with Processing. More info here: https://github.com/oxullo/Arduino-MAX30100/tree/master/extras/rolling_graph
The readout looks good to me, except for a totally saturated IR channel.

Try to change this line: https://github.com/oxullo/Arduino-MAX30100/blob/e63d7da5e7e202f425859648823b7501f9b37ee2/examples/MAX30100_RawData/MAX30100_RawData.ino#L33

To:

#define IR_LED_CURRENT                      MAX30100_LED_CURR_11MA

Try again the rolling graph, it'll give you insights on the readout.

oxullo, 2017-11-21:
@njordan77 could you try the new released version (1.2.0) altogether?
If it doesn't solve the issue, please open a new ticket and fill in the details of your setup and problem on the issue template.

oxullo, 2018-02-02:
Closing the issue due to inactivity

closed 2018-02-02 by oxullo