oxullo/Arduino-MAX30100 #30: only until the initialization can not be a failed or successful message

dimanes, 2017-11-04:
### Troubleshooting checklist

- [yes] I read the README (on master) thoroughly
- [yes] I ran the MAX30100_Tester and I'm going to paste the output down below
- [yes] I filled in all the details of my setup down below

### Description of the issue
i have a problem with my rcwl max30100, i have compiled according to wiring document. I use arduino uno, when I compile code test, minimal, and debug I only got intializing max30100 only. did not get a success or failed message. thanks maybe someone can help

### Output from MAX30100_Tester example
[image]

### Details of my setup

* Arduino hardware:  arduino uno
* MAX30100 breakout:
gnd - gnd
3.3v - vin
int - d2 and 4.7k 3.3v
rd and ird - 3.3v and gnd 0.1uF
scl - a5 and 4.7k 3.3v
sda - a4 and 4.7k 3.3v
* Arduino framework version: 1.8.5
* MAX30100 library version: new version

oxullo, 2017-11-19:
@dimanes I added some notes to the readme file, in particular regarding setups like yours (chapter "Logic level compatibility"). Please have a look and get back here afterwards. https://github.com/oxullo/Arduino-MAX30100/blob/master/README.md

misan, 2017-12-05:
Same here with an original Arduino UNO.

 However, it does work on an Arduino MEGA. 

I am using a MAX30100 module from eBay powered at 5V that includes a 2.8V regulator. 

No external pull-up resistors used (though I read your instructions for using 4.7K).  Because some other I2C modules worked flawlessly without adding them, many people are going to assume this one should too :-(

oxullo, 2017-12-10:
@misan when the breakouts include the pullups like the one you posted, there's no need for additional care, but as I see the three pullups are connected to the 1.8V rail instead of the 3.3V.
And even if it was connected to the 3.3V there would had been no certainty that it was going to work with a 5V MCU (logic level thresholds for cmos..).
So (@dimanes you too if it happens to have the same breakout) try to desolder the three pullups and add external pullups to the Vcc of the MCU. Consider that SCL, SDA, /INT of the MAX30100 safely handle up to 6V (page 2 of the datasheet).

oxullo, 2018-02-02:
Closing the issue due to inactivity

closed 2018-02-02 by oxullo