oxullo/Arduino-MAX30100 #32: Getting strange pulses each 2 seconds

bestware, 2017-11-18:
### Troubleshooting checklist

### Description of the issue
Using the debug-prog + rolling graph can see strange pulses occuring exactly every 2 seconds! 
(Very high peak followed by some negative samples) - then ok till next peak.

No idea where  this peaks come from (visible in Serial monitor as well) 
[image]

### Output from MAX30100_Tester example
Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=38.56C
All test pass. Press any key to go into sampling loop mode

### Details of my setup
* Arduino hardware:
Mega 2560 (different hardware tested - same result)
* MAX30100 breakout:
RCWL-0530
* Arduino framework version:
* MAX30100 library version:
1.10 (newest)

oxullo, 2017-11-19:
@bestware thanks for pointing it out. It's surely a regression caused by b2fa5675 and it could be also the reason for a bunch of reports regarding the unreliable bpm count.

oxullo, 2017-11-19:
@bestware could you give a try to the branch https://github.com/oxullo/Arduino-MAX30100/tree/issues/%2332/workaround ?

bestware, 2017-11-19:
Great - works like a charm!
Thank you for quick reply.
[image]

oxullo, 2017-11-19:
It looks to me that it's a bug of the MAX30100. The flag to start the temperature readout must be set on register 0x6, which controls the operating mode of the sensor and I guess that setting this register triggers a momentary shutdown of the LED drivers, injecting a transient in the samples series and temporarily blinding the bpm counter.
In the new design, MAX30102 the temperature enable flag has been silently moved to a different register..

I fixed it by removing the temperature poller from the PulseOximeter class. The change is in master and it will be released soon.

closed 2017-11-19 by oxullo