oxullo/Arduino-MAX30100 #33: Arduino Nano+MAX30100

AhmadZMQ, 2017-11-22:
**@oxullo**
 ### Troubleshooting checklist

- [ y] I read the README (on master) thoroughly
- [ y] I ran the MAX30100_Tester and I'm going to paste the output down below
- [y] I filled in all the details of my setup down below

### Description of the issue
RED led does not light up normally,It got lighten when it connects with GND individually(except IR led to the GNDi.e open).
Even I tried with 10k ohm same issue is happening.
### Output from MAX30100_Tester example
[image]

### Details of my setup
NANO SDA->MAX30100 SDA
NANO SCL->MAX30100 SCL
with4.7k ohm pullup to get function.
Input-5V
* Arduino hardware:
Nano
* MAX30100 breakout:
MAX30100
* Arduino framework version:
Arduino v1.6.12
* MAX30100 library version:
MAXlib30100-Ver 1.1.0
(This is your lib published on the Arduino lib)

AhmadZMQ, 2017-11-22:
After updadting the lib -
MAXlib30100-Ver 1.1.0 toMAXlib30100-Ver 1.2.0
 Same issue persist.
What i will do for -----

oxullo, 2017-11-22:
Could you post the schematics of your connections? and visually explain what you mean by connecting gnd "individually"

AhmadZMQ, 2017-11-23:
Thanks for your reply,Here I am attaching the screen shot of every connection with RD and IRD pin-
1.RD pin get lighten when it goes to the GND alone.connection is - /home/zmq171/Ahmad/AhmadDocuments/Spo2/RDPin ToGND.jpg
2.Both RD and IRD connect with GND,due to this the RED light doesn't got lighten.The connectio is-
/home/zmq171/Ahmad/AhmadDocuments/Spo2/RD_IRDBoth GND.jpg

**Note-**I have alredy check the voltage and current on the pins,This shows -
SDA-3.08v
SCL-3.08v
IRD-50mA
RD-11mA
Input -5V also with pullup of 4.7k ohm.

When the RED led on and I am placing the finger on it but doesn't happen any thing it still shows-
RED=0 IR=0
What we have to do ,Please help me .

AhmadZMQ, 2017-11-23:
[image]
[image]

AhmadZMQ, 2017-11-23:
This is not closed it is done by mistake.
Sorry.

AhmadZMQ, 2017-11-23:
This is happen by mistake so sorry.

AhmadZMQ, 2017-11-24:
Also a one more thing when i am changing the power supply of RED led their current always shows the 8mA irrespective of changing the voltage.how it can be solve.Please let me know.

nazmibojan, 2017-11-24:
@AhmadZMQ have you measured voltage input in Pin 11 (VDD) of MAX30100? is that 1.8V?

AhmadZMQ, 2017-11-24:
When I supplied 3.3v the red led doesn't light up.So.I am supply the input voltage as 5V.
only the pin MAX30100 has-
Vin->5v
SCL->3.24v with pullup
SDA->3.24v with pullup
INT(Interrupt)->pin2(nano)with pullup[optional]
IRD->50 mA
RED->10 mA
GND

pullup of 4.7k ohm sensor does work and red led get lighten.Without pullup sensor doesn't work.

When I run the MAX30100_Tester.ino,the result is-
[image]

temp show the zero value,it should be greater than zero.what i do please help me.......

oxullo, 2017-11-26:
@AhmadZMQ measure the voltage at SDA/SCL when the firmware is not running. Try to measure while keeping the reset button of the nano pressed and also by disconnecting the connection between nano and max breakout.
Which breakout are you using? Can you post a link to its schematics?
IRD and RD, if they refer to the cathodes (respectively pins 5 and 6 of the sensor) you should leave them floating. Check also this issue https://github.com/oxullo/Arduino-MAX30100/issues/13

AhmadZMQ, 2017-11-27:
**1. Which breakout are you using?**--you can see by click on below link
https://www.amazon.in/Generic-MAX30100-Oximeters-Development-Arduino/dp/B07427ZT6N?tag=googinhydr18418-21

**2. IRD and RD, if they refer to the cathodes (respectively pins 5 and 6 of the sensor) you should leave them floating.** -->I don't know in which mode they are ,How can I know?

**3. Measure the voltage at SDA/SCL when the firmware is not running**-->I upload the bare minimum(i.e. blank sketch )and measure the voltage,which is as-

**When - SDA ,SCL connected to the NANO.**
**a.** Without pullup-
SDA-->0.36v and SCL-->0.36v keep IRD and RED Led open(no connection).
**b.** With pullup-
SDA-->3.53v and SCL-->3.53v keep IRD and RED Led open(no connection).

**When - SDA ,SCL not connected to the NANO.(only connected through power and GND)**
**a.** Without pullup-
SDA-->0.00v and SCL-->0.00v constantly decreasing and stable at 0.00v. Keep IRD and RED Led open(no connection).
**b.** With pullup-
SDA-->3.53v and SCL-->3.53v keep IRD and RED Led open(no connection).

**4.Try to measure while keeping the reset button of the nano pressed and also by disconnecting the connection between nano and max breakout.**-->This is as--->
**a.** Without pullup-
SDA-->3.53v and SCL-->3.53v keep IRD and RED Led open(no connection).
**b.** With pullup-
SDA-->3.53v and SCL-->3.53v keep IRD and RED Led open(no connection).

INPUT voltage same -->5v
If I supplied 3.3v the voltage at the SDA SCL, value is 1v less i.e.2.24v with pullup.

I got very frustrating with it,I am facing this issue last more than 15 days.Please help me any how because I try every combination,connection ,current setting by playing with Library "current selection MACROS",Also by read the thoroughly datasheet of MAX30100 provided by MAXIM.

oxullo, 2017-12-10:
@AhmadZMQ check #30 you might have the same issue, even though it's weird that you recorded 0V on SDA/SCL with the power connected.
The voltage drop when you pull up the lines might be caused by a current path to ground or to the 1.8V VRU or due to a damaged I2C driver.

oxullo, 2018-02-02:
Closing the issue due to inactivity

closed 2018-02-02 by oxullo