oxullo/Arduino-MAX30100 #35: Compiling in Arduino IDE

vctrkarthik, 2017-11-28:
### Troubleshooting checklist

- [y ] I read the README (on master) thoroughly
- [ y] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ y] I filled in all the details of my setup down below

### Description of the issue
I copied all the files in src folder into MAX30100_Tester and verified the code. I get following error message.
### Output from MAX30100_Tester example
In file included from sketch\MAX30100.h:25:0,

                 from C:\Users\Victor\Downloads\Arduino-MAX30100-master\Arduino-MAX30100-master\examples\MAX30100_Tester\MAX30100_Tester.ino:24:

sketch\CircularBuffer.h:117:30: fatal error: CircularBuffer.tpp: No such file or directory

 #include <CircularBuffer.tpp>
### Details of my setup
Could you tell me how can I correct this? 

* Arduino hardware:
* MAX30100 breakout:
* Arduino framework version:
* MAX30100 library version:

oxullo, 2017-12-10:
I can't reproduce, I can compile properly. Please fill in the details of the versions used.

vctrkarthik, 2017-12-10:
Later, I used mikroelektronika max30100 development board. Then it worked.
thanks!!

closed 2017-12-10 by oxullo

leogaltieri, 2018-02-13:
hi i have the same problem, i am using arduino due with rcwl 0530
Description of the issue
I copied all the files in src folder into MAX30100_Tester and verified the code. I get following error message.

In file included from sketch\MAX30100.h:25:0,
from C:\Users\exo\Downloads\Arduino-MAX30100-master\Arduino-MAX30100-master\examples\MAX30100_Tester\MAX30100_Tester.ino:24:

sketch\CircularBuffer.h:117:30: fatal error: CircularBuffer.tpp: No such file or directory
#include <CircularBuffer.tpp>

you can help me please??

oxullo, 2018-02-18:
@leogaltieri please open a new issue and make sure you fill in the details required on the proposed description field