oxullo/Arduino-MAX30100 #36: i will ask if this result is right

eduard88806815, 2017-11-29:
### Troubleshooting checklist

- [ ] I read the README (on master) thoroughly
- [ ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ ] I filled in all the details of my setup down below

### Description of the issue

### Output from MAX30100_Tester example

### Details of my setup

* Arduino hardware:mega 2560
* MAX30100 breakout:connection VIN=>5V,SCL and SDA(with4.7k resistor)=>SCLandSDA, GND=>GND
* Arduino framework version:
* MAX30100 library version:

[image]

oxullo, 2017-12-10:
Have you tried https://github.com/oxullo/Arduino-MAX30100#sketchy-beat-frequency-readouts ?

oxullo, 2018-02-02:
Closing the issue due to inactivity

closed 2018-02-02 by oxullo