oxullo/Arduino-MAX30100 #41: Using two MAX30100 together in one Arduino

yiank, 2018-01-24:
Hello, 
Is it possible to use two MAX30100 together in one Arduino? If yes, how I will wire them?

oxullo, 2018-02-02:
Given the fact that MAX30100 has a fixed I2C address, you can't put them in a shared bus. Either use one hardware I2C bus and connect the second unit to arbitrary pins, instantiating a software I2C (eg: https://github.com/felias-fogg/SoftI2CMaster ). Anyway this is quite pointless, since the software bus will be too slow to allow the PulseOximeter class to operate properly and anyway the average computing power for a single PulseOximeter is almost capping a 16MHz ATMega328p.
Switch to a more powerful platform or simply use one micro per sensor

closed 2018-02-02 by oxullo

yiank, 2018-02-05:
Thank you for your reply. So, if I use and arduino due that has 2 I2C buses, how I define witch bus each sensor will use?