oxullo/Arduino-MAX30100 #42: Node MCU and max30100

JanashreeA, 2018-02-05:
I am trying to connect my max30100 with my nodeMCU
I am getting the following error when i tried to compile the example MAX30100_Tester.ino

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp: In member function 'bool BeatDetector::checkForBeat(float)':

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:64:67: error: no matching function for call to 'min(float&, int)'

                 threshold = min(sample, BEATDETECTOR_MAX_THRESHOLD);

                                                                   ^

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:64:67: note: candidates are:

In file included from c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\ANAND\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:19:

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:64:67: note:   mismatched types 'std::initializer_list<_Tp>' and 'float'

                 threshold = min(sample, BEATDETECTOR_MAX_THRESHOLD);

                                                                   ^

In file included from c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\ANAND\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:19:

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:64:67: note:   mismatched types 'std::initializer_list<_Tp>' and 'float'

                 threshold = min(sample, BEATDETECTOR_MAX_THRESHOLD);

                                                                   ^

In file included from c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\ANAND\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:19:

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:64:67: note:   deduced conflicting types for parameter 'const _Tp' ('float' and 'int')

                 threshold = min(sample, BEATDETECTOR_MAX_THRESHOLD);
[... 93 lines cut ...]
                 from C:\Users\ANAND\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/Arduino.h:240,

                 from C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:19:

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\anand\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100\src\MAX30100_BeatDetector.cpp:81:67: note:   deduced conflicting types for parameter 'const _Tp' ('float' and 'int')

                 threshold = min(sample, BEATDETECTOR_MAX_THRESHOLD);

                                                                   ^

Using library Wire at version 1.0 in folder: C:\Users\ANAND\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Wire 
Using library Arduino-MAX30100 at version 1.2.0 in folder: C:\Program Files (x86)\Arduino\libraries\Arduino-MAX30100 
exit status 1
Error compiling for board Generic ESP8266 Module.

Please do help
Thanks in advance

oxullo, 2018-02-05:
@JanashreeA the library doesn't officially support nodemcu, but in order to solve your problem, I guess you should just add an implementation of the min() function:

```
#define min(a,b) \
   ({ __typeof__ (a) _a = (a); \
       __typeof__ (b) _b = (b); \
     _a < _b ? _a : _b; })
```
To the file MAX30100_BeatDetector.cpp
The compiler must support the typeof operator.

closed 2018-02-05 by oxullo

JanashreeA, 2018-02-06:
Thank You so much...........................my max30100 works
Thanks a lot

JanashreeA, 2018-02-24:
@esz11 
I did not use 4.7k pull up bcoz i used RCWL - 0530 (max30100 breakout) that has internall pull ups
Vin - 3.3 v
Gnd -Gnd
SCL-D1
SDA-D2

then make the min def in the MAX30100_BeatDetector.cpp
thats all 
If u have done this then your  Max30100 will work perfectly
First dump the MAX30100_Tester then MAX30100_rawdata and then MAX30100_minimal in the examples folder
It will work

esz11, 2018-02-25:
@JanashreeA 
Thank you so much. It is working successfully.

proyectoTEOS, 2018-03-12:
Hi guys, should I create a function called min?

void min(){
     #define min(a,b) \ ({ __typeof__ (a) _a = (a); \ __typeof__ (b) _b = (b); \ _a < _b ? _a : _b; }) 
}

or I just have to declare the line of code after:

#include <Arduino.h>
#include "MAX30100_BeatDetector.h"

#define min(a,b) \ ({ __typeof__ (a) _a = (a); \ __typeof__ (b) _b = (b); \ _a < _b ? _a : _b; }) 

BeatDetector::BeatDetector() :
    state(BEATDETECTOR_STATE_INIT)..... etc

I still get errors, thanks for your support.

oxullo, 2018-03-12:
it should be fixed in the latest release v1.2.1

proyectoTEOS, 2018-03-13:
I have this problem in the serial port:

Soft WDT reset

ctx: cont 
sp: 3ffefbf0 end: 3ffefdd0 offset: 01b0

>>>stack>>>
3ffefda0:  feefeffe 3ffeea7c 3ffeed80 40202570  
3ffefdb0:  feefeffe 00000000 3ffeed9c 40203bac  
3ffefdc0:  feefeffe feefeffe 3ffeedb0 40100108  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
Initializing pulse oximeter..FAILED

JanashreeA, 2018-03-13:
@proyectoTEOS 
It is a software watchdog timer problem
Software reset was triggered 
To see which line triggers the wdt just install ESP Exception decoder
See the below link for more information 
arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.htmlthis site is awesome

Before that after uploading the code just unplg your nodemcu and then plug it to power 
Because this may solve your problem easily ....since i do this first before trying other solution

I hope you got it.......if you had any doubt comment your doubt
Thank you

proyectoTEOS, 2018-03-13:
Hello, a final problem.

I have several sensors connected to my Node MCU, including the MAX30100 and a DS18B20. The problem is that they work at different speeds, this generates a conflict with the MAX30100 sensor since it does not detect the changes. I put here a short version of the code, but it still does not update the value of the pulse / oximeter sensor.

#include <Wire.h>
#include "MAX30100_PulseOximeter.h"
#include <OneWire.h>
#include <DallasTemperature.h>
#define REPORTING_PERIOD_MS 1000
#define REPORTING_PERIOD_MS2 2000

const int sensorPinT = D6;

PulseOximeter pox;

OneWire wireT(sensorPinT);
DallasTemperature dallasT(&wireT);

uint32_t tsLastReport = 0;
uint32_t tsLastReport2 = 0;

void setup() {
  dallasT.begin();
  Serial.begin(115200);
  Serial.print("Initializing pulse oximeter..");
  if (!pox.begin()) {
    Serial.println("FAILED");
    for (;;);
  } else {
    Serial.println("SUCCESS");
  }
}

void loop() {
  pox.update();
  dallasT.requestTemperatures();

  if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
    Serial.print("Heart rate:");
    Serial.print(pox.getHeartRate());
    Serial.print("bpm / SpO2:");
    Serial.print(pox.getSpO2());
    Serial.println("%");
    tsLastReport = millis();
  }

  if (millis() - tsLastReport2 > REPORTING_PERIOD_MS2) {
    Serial.print("Temperatura actual: ");
    Serial.print(dallasT.getTempCByIndex(0));
    Serial.print(" GC");
    Serial.print("  ||  ");
    Serial.print(dallasT.getTempFByIndex(0));
    Serial.println(" GF");
    tsLastReport2 = millis();
  }
}

JanashreeA, 2018-03-14:
@proyectoTEOS 
Maxso100 uses asynchronous call back and pox.update() must be called as soon as possible
I too work with many sensors and I too faced the same software reset problem
So to avoid it I just added pox.begin() and pox.update in middle of the code where it blocks the esp8266 from calling its utility function i.e where there is large delay to perform the operation

etherFlare, 2018-04-15:
@proyectoTEOS  
hello can youi tell me how did you fix this problem
I have this problem in the serial port:

Soft WDT reset

ctx: cont
sp: 3ffefbf0 end: 3ffefdd0 offset: 01b0

stack>>>
3ffefda0: feefeffe 3ffeea7c 3ffeed80 40202570
3ffefdb0: feefeffe 00000000 3ffeed9c 40203bac
3ffefdc0: feefeffe feefeffe 3ffeedb0 40100108
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
Initializing pulse oximeter..FAILED

i'm having the same problem . can u help me please .. thanks in advance

proyectoTEOS, 2018-05-18:
Hey, @alrawajbeh96 my code is in a couple of comments above, I can not solve my problem

za1nabahmed, 2019-05-28:
plase this my problem with MAX30100 how can i solve it ?
>>>stack>>>
3ffefc60:  3fffdad0 3ffeeb1c 3ffeec40 402028c8  
3ffefc70:  feefeffe 00000000 3ffeec5d 40203a8c  
3ffefc80:  feefeffe feefeffe 3ffeec70 40100958  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,4)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Initializing pulse oximeter..FAILED

kentpaul, 2020-04-24:
hello sir did you solve that problem? im having the same problem here please help