oxullo/Arduino-MAX30100 #45: Error on compile!

Boubis, 2018-03-07:
hello! Every example i try to compile i always get this error about a specific header file!
i have all the libraries and header files neccesery in the same folder as the .ino file.
[image]

Can you help me?

oxullo, 2018-03-08:
Please submit a proper issue report indicating all the required info

closed 2018-03-08 by oxullo

Boubis, 2018-03-08:
i cant understand a more simple way to explain it!
when i try to compile MAX30100_Minimal example there is a compile error!
here is the exact error message:
[image]

i have all the nessecery header files and libraries in the same folder 
[image]

[image]

i hope i make myself more inambiguous! Your help is very valuable

oxullo, 2018-03-08:
When you create an issue, there's a default text on the body of the issue that requires you to fill in the details of the environment and circumstances. I can help if you help me first.

Boubis, 2018-03-08:
sorry i am new to github and i dont know! you have to answer only to this format that you use? or "unofficially" you can make an exception and answer here? If you cant i understand and i will create an issue properly