oxullo/Arduino-MAX30100 #46: Error on compile

Boubis, 2018-03-08:
### Troubleshooting checklist

- [yes ] I read the README (on master) thoroughly
- [ yes] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ yes] I filled in all the details of my setup down below

### Description of the issue
CirculaBuffer.tpp is a file that exists in the directory but there is some issue and it cant link with the Tester.
maybe .tpp extension is wrong?

### Output from MAX30100_Tester example
[image]

### Details of my setup

i dont think this part matters because the code doesnt pass the compile phase

* Arduino hardware: Uno 
Vdd ->5V
Gnd->Gnd
SDA->SDA
SCL->SCL
INT->Pin 2

* MAX30100 breakout: rcwl-0530
* Arduino framework version: -
* MAX30100 library version: -

Boubis, 2018-03-09:
IDE version 1.8.1
i use the latest version 1.2.0

per1234, 2018-03-09:
There are actually two issues here:

1. @Boubis has the library files in the sketch folder and the library incorrectly uses angle braces syntax for a local file in CircularBuffer.h, line 117:
`#include <CircularBuffer.tpp>`
Although wrong, that works fine when the library is installed in a normal location but it does not work when the library is located outside the Arduino IDE's default #include search path, such as in the sketch folder.
2. Even though it works fine when the library is installed normally, the .tpp extension apparently is not supported by the Arduino IDE for files in sketch folders.

If I correct the `#include` syntax and change the filename of CircularBuffer.tpp to a supported extension then the problem is solved.

So you might say "Well, just install the library normally. Don't put it in the sketch folder" but actually it can be very useful to be able to bundle all 3rd party library dependencies of a sketch in the sketch folder so that it can be distributed as a self contained package.

oxullo, 2018-03-09:
@per1234 thank you very much for your detailed analysis.

@Boubis if you confirm that the issue happens only when you bundle the library into your sketch folder, my suggestion is to simply paste the content of CircularBuffer.tpp into the CircularBuffer.h replacing the include and deleting the former, or renaming it to CircularBuffer_Template.h and adjust the include statement accordingly, as @per1234 suggests.

Another option, which you may take just a suggestion, is to use [platformio](https://platformio.org/) instead of Arduino IDE.

Boubis, 2018-03-10:
@per1234  and @oxullo thank you for all the help so far! i understand the issue and i fix it! Now when i run Tester i have this output on serial 
[image]
My wiring is this:
[image]
Vin -> 5V
Gnd -> Gnd
SCL -> SCL (4,7K pullup)
SDA -> SDA (4,7K pullup)
INT -> INT (4,7K pullup)
IRD and RD floating

oxullo, 2018-03-10:
As the tester hints, you're using the wrong part. As I see from the picture, it looks like a MAX30101 or MAX30102, which aren't compatible with the library.

Boubis, 2018-03-10:
[image]
the breakout board says MAX30100

oxullo, 2018-03-10:
The "RCL" breakouts are pesky. Poorly designed, I2C pullups to the 1V8 line and now I see that they're also lying on the actual sensor..
This is a MAX30100: https://www.mouser.com/images/microsites/Maxim_MAX30100.jpg
And this a MAX30102: https://www.mouser.com/images/microsites/maxim%20MAX30102EFD_t.jpg
You can notice the increased size of the sensor window.

MAX30100 are out of production since a while.

oxullo, 2018-03-10:
Another annoying thing from Maxim is that MAX30101 and MAX30102, very similar yet two different models, report the same product id (0x15) when querying register 0xff. But what you have is definitely one of the two.

Boubis, 2018-03-10:
I understand and i am so disappointed...
I think that this library cant be modified to work for that module right?

oxullo, 2018-03-10:
Yes and no.. There are some backward-incompatible rearrangements of the registers, but the only part that needs to be partially re-written is the low-level device driver (MAX30100 class and its registers). I actually wanted to add support to the new chips, but finding time is always complicated :)

Boubis, 2018-03-10:
I fully understand! i will try one my own and i will sent if i have any update!
Anyway thank you @oxullo, @per1234 for all your time and valuable advices!

oxullo, 2018-03-10:
Good luck!

closed 2018-03-10 by oxullo

kanaksony, 2022-07-02:
> There are actually two issues here:
>
> 1. @boubis has the library files in the sketch folder and the library incorrectly uses angle braces syntax for a local file in CircularBuffer.h, line 117: [...]

I was trying the example code "MAX30100_Minimal.ino", I kept all library src code in parallel the .ino file, which was throwing error while compiling it.
The error looked like -
`In file included from E:\Data_from_Huawei_Lenovo\Downloads\BITS\Course Content\Course 5\Oxymeter\Arduino-MAX30100-master\examples\MAX30100_Minimal\MAX30100.h:25:0,
                 from E:\Data_from_Huawei_Lenovo\Downloads\BITS\Course Content\Course 5\Oxymeter\Arduino-MAX30100-master\examples\MAX30100_Minimal\MAX30100_PulseOximeter.h:30,
                 from E:\Data_from_Huawei_Lenovo\Downloads\BITS\Course Content\Course 5\Oxymeter\Arduino-MAX30100-master\examples\MAX30100_Minimal\MAX30100_Minimal.ino:20:
CircularBuffer.h:117:10: fatal error: CircularBuffer.tpp: No such file or directory
 #include "CircularBuffer.tpp"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Using library Wire at version 1.0 in folder: C:\Users\user\Documents\ArduinoData\packages\arduino\hardware\avr\1.8.5\libraries\Wire 
exit status 1
CircularBuffer.tpp: No such file or directory`

As per suggested by @per1234, I removed src of library for project folder and imported it from Managed libraries in Arduino iDE, and it compiled perfectly.

The reason Arduino IDE doesn't recognizes .tpp files.

Thanks,
Kanak