oxullo/Arduino-MAX30100 #49: Sensor integration, the MAX sensor value is not updated.

proyectoTEOS, 2018-03-13:
Hello, a final problem.

I have several sensors connected to my Node MCU, including the MAX30100 and a DS18B20. The problem is that they work at different speeds, this generates a conflict with the MAX30100 sensor since it does not detect the changes. I put here a short version of the code, but it still does not update the value of the pulse / oximeter sensor.

#include <Wire.h>
#include "MAX30100_PulseOximeter.h"
#include <OneWire.h>
#include <DallasTemperature.h>
#define REPORTING_PERIOD_MS 1000
#define REPORTING_PERIOD_MS2 2000

const int sensorPinT = D6;

PulseOximeter pox;

OneWire wireT(sensorPinT);
DallasTemperature dallasT(&wireT);

uint32_t tsLastReport = 0;
uint32_t tsLastReport2 = 0;

void setup() {
dallasT.begin();
Serial.begin(115200);
Serial.print("Initializing pulse oximeter..");
if (!pox.begin()) {
Serial.println("FAILED");
for (;;);
} else {
Serial.println("SUCCESS");
}
}

void loop() {
pox.update();
dallasT.requestTemperatures();

if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
Serial.print("Heart rate:");
Serial.print(pox.getHeartRate());
Serial.print("bpm / SpO2:");
Serial.print(pox.getSpO2());
Serial.println("%");
tsLastReport = millis();
}

if (millis() - tsLastReport2 > REPORTING_PERIOD_MS2) {
Serial.print("Temperatura actual: ");
Serial.print(dallasT.getTempCByIndex(0));
Serial.print(" GC");
Serial.print(" || ");
Serial.print(dallasT.getTempFByIndex(0));
Serial.println(" GF");
tsLastReport2 = millis();
}
}

oxullo, 2018-03-25:
The library requires very strict timing updates for its goal of 100Hz sampling rate. Delays or jitter would surely cause averaging and filtering issues that will reflect on the outcome.
Use the grapher app to visually inspect the quality of the signal, this will give you a rough idea of the nature of the jitter.
Profile your code and make sure that pox.update() is run less than every 10ms.
You can adjust the sampling frequency but this requires also to recalculate the FIR coefficients for the filters.

boyiddha, 2022-03-27:
after struggling few days, I read this blog:  
https://forum.arduino.cc/t/arduino-code-max30100-and-ds18b20-doesnt-work/899066/10
and I also read many blogs then I got something like this:
I used
 sensors.setWaitForConversion(false);            //as a result, requestTemperatures() will not block current thread
before this command
sensors.requestTemperatures(); 
.
.
In my code, above two lines look like this:

sensors.setWaitForConversion(false); 
sensors.requestTemperatures(); 
.
.
and it works!

pavanambat, 2022-10-17:
> after struggling few days, I read this blog: https://forum.arduino.cc/t/arduino-code-max30100-and-ds18b20-doesnt-work/899066 and I also read many blogs then I got something like this: I used sensors.setWaitForConversion(false); //as a result, requestTemperatures() will not block current thread before this command sensors.requestTemperatures(); and after above command : sensors.setWaitForConversion(True); // sync for next conversion . . In my code, above three lines look like this:
> 
> sensors.setWaitForConversion(false); sensors.requestTemperatures(); sensors.setWaitForConversion(True); . . and it works!

Bro can u send full code

Dimzz44, 2023-03-08:
> Bro can u send full code

have u done it? i have a same problem rn

boyiddha, 2023-03-08:
@pavanambat 
@Dimzz44 
My project Github link: [https://github.com/boyiddha/IoT-Based-Patient-Health-Monitoring-System](url)
.
Also, check the below requirement-
•	Poor quality USB cable;
•	The USB cable is too long;
•	Board with some defect (bad solder joints);
•	Bad computer USB port;
•	Or not enough power provided by the computer USB port.
Fix: Try with a different shorter USB cable (with data wires), try a different computer USB port, or use a USB hub with an external power supply.

boyiddha, 2023-03-08:
@pavanambat
@Dimzz44
My project Github link: [github.com/IoT Based Patient Health Monitoring System](https://github.com/boyiddha/IoT-Based-Patient-Health-Monitoring-System)
.
Also, check the below requirement-
• Poor quality USB cable;
• The USB cable is too long;
• Board with some defect (bad solder joints);
• Bad computer USB port;
• Or not enough power provided by the computer USB port.
Fix: Try with a different shorter USB cable (with data wires), try a different computer USB port, or use a USB hub with an external power supply.