oxullo/Arduino-MAX30100 #51: max 30100(rcwl0530) not work

dianjach, 2018-03-25:
I'm having problems. max30100 (rcwl0530) can not work. I'm using your library max30100. and run the max30100 tester and just show it
[image]
[image]
[image]
[image]

pin max30100 vin = pin arduino v3,3
pin max30100 scl = pin arduino scl ( 4,7 R )
pin max30100 sda = pin arduino sda  ( 4,7 R )
pin max30100 int = pin arduino d2  ( 4,7 R )
pin max30100 GND = pin arduino GND

can you help me ??? :(

oxullo, 2018-03-25:
@dianjach  did you read the README? You can find some info about the crappy RCWL-0530 there. Consider though that this breakout is poorly designed.

dianjach, 2018-03-26:
"the three 4k7 pullups must be desoldered and external 4.7k pullups to Vcc of the MCU must be added". how to use this. I do not understand :(. can you explain ??

tuttelikz, 2018-04-02:
No need for voltage regulator, capacitors. They are already integrated into rcwl0530.
Here is the wiring that worked for me:

Arduino -> MAX30100 (rcwl0530):
vin -> 3.3V
gnd -> gnd
scl -> scl (4.7 kOhm parallel pullup)
sdl -> sdl (4.7 kOhm parallel pullup)
id -> gnd (4.7 kOhm series) to reduce current
ird -> gnd (4.7 kOhm seris) to reduce current

dianjach, 2018-04-08:
thanks all, max30100 can work.
I want to ask, how to read sensor max30100 to 5 seconds?
I think the sensor reading is too fast.
Please help me :)
thanks

tuttelikz, 2018-04-09:
Sensor does not show correct values. It is around 30-40 bpm. Anyone had this issue? Maybe library issue?

tuttelikz, 2018-04-09:
sensor value is jumping with big range. probably i need to average out frequency of detected beats and calibrate from empirical values.

dianjach, 2018-04-09:
yes, i also have problem sensor data. the sensor value jumps with a large range. how is the data sensor stable ?? anyone can help

tuttelikz, 2018-04-09:
@dianjach,

I tried to average after 3000 iterations. It is still jumping.

i checked MAX30100_BeatDetector.cpp
There getRate (heart rate) is defined as:
pulse (bpm) = (1/T) * 1000 * 60

When I tried to receive that T (period), it changes much.
So, I think the problem is with detection of beats.

Some of possible ways we can try is to:
- find out some threshold values and adjust them (if there are any in the library)
- consider noise canceling or lowpass

If someone has experience, please share :)

tuttelikz, 2018-04-10:
Hey @dianjach, seems got a solution :)

Try to run this arduino library
https://github.com/xcoder123/MAX30100

Here is a plot:
[image]

dianjach, 2018-04-10:
thank you @tuttelikz  , I will try it.
if using that library, what wiring sensor to arduino is changed?

tuttelikz, 2018-04-10:
@dianjach, wiring is same :)

keilanigrace, 2018-04-10:
Hello, this is my first time on github so please forgive me for any mistakes. I had the same problem as @dianjach and changed my wiring to match @tuttelikz suggestion, but it's still showing Initializing MAX30100...
Here is my wiring. Could someone please help me figure out what might be wrong? 
[image]

[image]

tuttelikz, 2018-04-10:
sorry @keilanigrace . difficult to see all the connections from the picture.

azlinasuerman, 2018-04-18:
@dianjach can I have your email ? I  have something to ask Thank you

fedy0, 2018-04-20:
@dianjach what worked for you? 
Is it exactly as @tuttelikz described in his comments below 
"...Arduino -> MAX30100 (rcwl0530):
vin -> 3.3V
gnd -> gnd
scl -> scl (4.7 kOhm parallel pullup)
sdl -> sdl (4.7 kOhm parallel pullup)
id -> gnd (4.7 kOhm series) to reduce current
ird -> gnd (4.7 kOhm seris) to reduce current"
I'd like to know cos I am facing same challenge even with @keilanigrace

Can someone explicitly summarizes how to tackle this problem? @oxullo Your input

3alarm, 2018-05-23:
hanmanuk !!!! I love you!
I cant believe i could solder this tiny thing, but you are right!  and now it works!!
Thank you!

qyomk, 2018-07-26:
I am using MAX30100_RCWL-0530 with Particle Photon, my connections are (3V3 -> VIN, SDA -> D0, SCL -> D1, GND-> GND) ,INT, IRD, RD not connected to photon, i am trying to the run MAX30100 library example, I got wrong data

Beat!
Heart rate:29bpm / Spo2:0%
Heart rate:29bpm / Spo2:0%
Heart rate:29bpm / Spo2:0%

What should i do to fix this issue ??

3alarm, 2018-07-26:
that breakout board is wired wrong, you need to jump a wire and scratch off some of the printed circuit, i would highly recommend getting the more recent version of that particle sensor from spark fun.. i think that is where i got it.it looks like thisNew video by Kevin Leeser

  
|  
|   
|   
|   |    |

   |

  |
|  
|    |  
New video by Kevin Leeser
   |   |

  |

  |

3alarm, 2018-07-26:
SparkFun Particle Sensor Breakout - MAX30105
  
|  
|   
|   
|   $ 12.95   |    |

   |

  |
|  
|   |  
SparkFun Particle Sensor Breakout - MAX30105
 The SparkFun MAX30105 Particle Sensor is a flexible, powerful sensor enabling sensing of distance, heart rate, p...  |   |

  |

  |

 
 

    On Thursday, July 26, 2018 4:46 PM, Kevin Leeser <3alarm@ameritech.net> wrote:
 

 that breakout board is wired wrong, you need to jump a wire and scratch off some of the printed circuit, i would highly recommend getting the more recent version of that particle sensor from spark fun.. i think that is where i got it.it looks like thisNew video by Kevin Leeser

  
|  
|  
|  
|   |    |

  |

  |
|  
|    |  
New video by Kevin Leeser
   |   |

  |

  |

Jivansha, 2019-01-02:
Hi, I am unable to solve the initial issue of only output on the serial monitor as "Initializing MAX30100.." while running the tester program. The initial connections were - 
Arduino -> MAX30100 (rcwl0530):
vin -> 3.3V
gnd -> gnd
scl -> scl (4.7 kOhm parallel pullup connected to 3.3v)
sdl -> sdl (4.7 kOhm parallel pullup connected to 3.3v)

I also tried the connections used by @tuttelikz  but no change in the output occurs. Is there any other way besides modifying the hardware design? Please share. 
Thanks

mguo46, 2019-05-07:
> I'm having problems. max30100 (rcwl0530) can not work. I'm using your library max30100. and run the max30100 tester and just show it
> [image]
> [image]
> [image]
> [image]
>
> pin max30100 vin = pin arduino [...]

mguo46, 2019-05-07:
I encounter same problem as dianjach!!!
 I'm using your library max30100. and run the max30100 tester and just show it
> [image]
> [image]
> [image]
> [image]
> 
> pin max30100 vin = pin arduino v3,3
> pin max30100 scl = pin arduino scl ( 4,7 R )
> pin max30100 sda = pin arduino sda ( 4,7 R )
> pin max30100 int = pin arduino d2 ( 4,7 R )
> pin max30100 GND = pin arduino GND
> 
> can you help me ??? :(

FelipeMdeO, 2019-05-07:
Are you using 4.7 R resistor?
You should use 4.7k.
Do it and try, plx

mguo46, 2019-05-07:
@FelipeMdeO  yes, I used 4.7K , sorry, wrong typing.

sarath341, 2019-12-13:
I Connected Vin of Sensor to 5V and it solved my Arduino Nano from resetting (Port Keeps Disconnecting)

arsa281197, 2020-06-01:
I am use it
vin -> 3.3V
gnd -> gnd
scl -> scl (4.7 kOhm parallel pullup connected to 3.3v)
sdl -> sdl (4.7 kOhm parallel pullup connected to 3.3v)
but not work "Initializing MAX30100..FAILED: I2C error"
can you help me please

humza-13, 2020-07-03:
Hey i am facing the same issue what i did on test run was i removed smd resistors and then i connected as follows
Arduino -> max30100
Vin -> 5v
GND-> GND 
scl -> a5 with 4.7k resistor 
Sda -> a4 with 4.7k resistor 
Int -> d2 with 4.7k resistor 
It was working fine on uno board when i transfer my design on nano it suddenly stopped working and now sensor is not blinking and my serial monitor says intializing but it never starts

vaishnavi160, 2021-01-23:
does anyone facing problem of sensor getting extremely hot ??

ElectronicsInnovation, 2021-03-18:
This video might help you, in this video, an Indian guy explained the issue in detail with the help of the max30100 sensor schematic diagram, and solved it in two ways.
https://youtu.be/ZqdmA4NAqb0
[image]