oxullo/Arduino-MAX30100 #54: Max30100 with Thingspeak

Marinamyat24, 2018-07-08:
### Troubleshooting checklist

- [OK ] I read the README (on master) thoroughly
- [OK ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [OK ] I filled in all the details of my setup down below

### Description of the issue
I am using Max30100 and the sensor reads the data correctly and bpm and spo2 values are correct  before uploading to thingspeak. After uploading to thingspeak, bpm and spo2 values only show "0".

### Output from MAX30100_Tester example
[image]

### Details of my setup
The wiring is as below:
MAX30100   --->   Arduino
     VIN                       5V
     SCL                      A5 with 4.7k ohm pullup resistor 
     SDA                     A4 with 4.7k ohm pullup resistor         
     INT                      3.3V
    GND                     GND
       

* Arduino hardware:   Arduino UNO
* MAX30100 breakout:  RCWL-0530
* Arduino framework version:  Arduino 1.8.5  
* MAX30100 library version:   MAX30100lib 1.2.1

pabloesya, 2019-01-04:
do you have some progress..... cause I'm in the same problem :'c

oxullo, 2019-02-04:
Please check #52

closed 2019-02-04 by oxullo