oxullo/Arduino-MAX30100 #56: Stuck at "Initializing pulse oximeter"

RinoHandarta, 2019-02-04:
Troubleshooting checklist
[OK] I read the README (on master) thoroughly
[OK] I ran the MAX30100_Tester and I'm going to paste the output down below
[OK] I filled in all the details of my setup down below
Description of the issue
Hello,
I just run MAX30100_Tester and get stuck in the serial monitor at "Initializing pulse oximeter.."
Why did that happen?
My MAX30100 - Arduino Wiring:
VIN -> 3.3V
GND -> GND
SDA -> SDA
SCL -> SCL
INT -> D2
[image]

Regards,
Rino
max30100 stuck

The output from MAX30100_Tester example
Initializing MAX30100

Details of my setup
Arduino hardware: Arduino Uno R3
MAX30100 breakout: https://www.tokopedia.com/caravane/max30100-pulse-oximeter-heart-rate-sensor-module
Arduino framework version: Arduino IDE 1.8.4
MAX30100 library version: v1.1.0

oxullo, 2019-02-04:
@RinoHandarta this is likely to be an I2C communication issue. Please doublecheck the README for some hints about the troubleshooting. Watch out also for the breakout board you're using, some are badly designed, in particular, check for the pullups resistors to SCL and SDA. Plus, it's likely that you're running an atmega328p at 5V with the I2C lines pulled up to 3V3: this *might* work but it's usually not recommended.

oxullo, 2019-02-16:
Closing due to inactivity. Feel free to reopen

closed 2019-02-16 by oxullo

Sv443, 2020-03-03:
I'm having the same issue and I'm currently looking for a solution.  
I found [this answer](https://github.com/oxullo/Arduino-MAX30100/issues/27#issuecomment-330160357) in issue #27 but I want to know if there's another way except buying an I<sup>2</sup>C level shifter module and waiting for it to ship.  
Any other ideas why this may be caused? I've already checked the wires' resistance, which is completely fine.