oxullo/Arduino-MAX30100 #57: MAX30100 (RCWL-0530) detects even without finger on

patty-victor, 2019-03-05:
### Troubleshooting checklist

- [X] I read the README (on master) thoroughly
- [X] I ran the MAX30100_Tester and I'm going to paste the output down below
- [X] I filled in all the details of my setup down below

### Description of the issue
I'm having some weird outputs for my sensor. It detects **anything**, even if I don't put my finger on it. The values it shows seem randomic. Whether I put my finger on it or not, the values keep varying in a very awkward range. I have the RCWL-0530 module, in which I **didn't** removed the embedded pullup 4k7 resistors, because I had another identical board in which I did that, and it just started to fail even to communicate through the I2C. 

Outside the module, I have 3k3ohms pullup resistors in SDA and SCL (I already changed these resistors to 4k7ohms with the same result), although I left a 4k7 ohms resistor pullup in INT. I'm using a Arduino UNO

My connections are: 
Vin -> 3.3V
GND -> GND
IRD and RD -> GND
Int -> Pin2 (4k7 pullup resistor)
SCL -> A5 (3k3 or 4k7 pullup resistor)
SDA -> A4 (3k3 or 4k7 pullup resistor)

### Output from MAX30100_Tester example

> Initializing MAX30100..Success
> Enabling HR/SPO2 mode..done.
> Configuring LEDs biases to 50mA..done.
> Lowering the current to 7.6mA..done.
> Shutting down..done.
> Resuming normal operation..done.
> Sampling die temperature..done, temp=0.00C
> WARNING: Temperature probe reported an odd value
> 
> Press any key to go into sampling loop mode

### Details of my setup

* Arduino hardware: UNO
* MAX30100 breakout: RCWL-0530
* Arduino framework version: Arduino IDE 1.8.8
* MAX30100 library version: latest

oxullo, 2019-03-06:
Hello @patty-victor, thanks for the detailed report. Most of the times the issue with this infamous breakout board is caused by getting out of specs for the i2c bus.
I'd suggest: 1. use 5V as Vin, this won't change much but at least you might prevent the onboard 3V3 LDO to drop a couple of hundred millivolts on the 3V3 rail. 2. desolder the three onboard 4k7 resistors (or sever their connection to the 1V8 rail)
Once done, keep your 3k3 or 4k7 pullups to 3V3. This is still out of specs, since the uno operates at 5V. If this doesn't work, then pull up SDA and SCL to 5V.
Make also sure that the sensor window is clean. Use IPA to ensure it's free of deposits.
If nothing work, it might be you have a defective unit.

patty-victor, 2019-03-06:
Hello OXullo, I’ll try your suggestions tomorrow evening, since today I’m not with my boards. Please, keep this issue open and I’ll post my outputs here as soon as I get them tomorrow. Thank you.

Outlook für iOS<https://aka.ms/o0ukef> beziehen

patty-victor, 2019-03-08:
Hi OXullo. I tried your tipps, here are my outcomes:
1- When I connected the VIN in the 5V, the Red LED started to blink after a while, as it was not doing when connected to 3V3 (it was constantly on). The output was even worse, as in 5V it doesn't even gathered any data. Output below.

[image]

2- Very reluctantly, I removed the three pullups resistors of the breakout board and supplied it on 3V3 (pullups also on 3V3) and this time it still communicated thought I2C! **VICTORY**! But the output was no much better, tho! The result was pretty much the same as the first scenario, but this time, when I started to clap my finger over the sensor very quickly, it started to detect something (something odd, but still something). The output below shows it. The first half is finger on, but still. The second, is finger clapping on sensor. I have no idea why it has such behavior.

[image]
 
3- My last try was supplying Vin with 5V and all 4k7ohms pullups also in 5V. It just failed I2C. The Tester script output is below. With 3V3 Vin and pullups, I have at least the weird output. Everything on 5V, and not even I2C working.

[image]

4- In my very last try, I supplied the Vin with 3V3 and the pullups on 5V. The result was the very same of step (2). Getting results only on clapping finger over the sensor. As below. I held still, started to clap, held still again, started to clap. And this was the output.

[image]

I noticed also that the sensor is getting gradually too hot to keep the finger on it, is that normal?
Otherwise, I'm quite out of ideas here :(

So this is it, sorry for the huge report, but I'm a bit hopeless, can you help me with some ideas? Thank you for the awesome library!

oxullo, 2019-03-09:
Hello @patty-victor 
I start from the bottom: I've never recorded a sensor getting noticeably warm, so I suspect there's something wrong with your setup. Do you observe such behavior with a specific configuration (eg: vin 5 or 3.3V)? Could it be that you connected any of the IRD or RD to ground? (don't! you'll kill the LEDs if it didn't happen already)

Check if the module you have matches the one in the site I use as reference: https://pekolibrary.wordpress.com/2018/05/27/rcwl-0530-max30100/

With the premises of VIN=5V and SCL/SDA pulled up to 5V with 4k7 (indulge me even if this setup doesn't work to you) check:
* VIN (=5V)
* SCL, SDA (=5V when you keep the arduino uno's reset button pressed)
* Pin 1 of the 3V3 LDO (=3.3V)
* Pin 3 of the 1V8 LDO (=1.8V)
* Make sure the pins marked IRD and RD are left floating (eg: not connected to anything)

Then disconnect SCL and SDA from the uno, you should read 0V there (supposing you haven't repopulated the onboard pullups).

Disconnect the power and check continuity between your known ground and pin 4 of the MAX30100.
Do the same with pin 12, which on the schematics is left floating (it should be connected to ground).

Remember to crosscheck your data with: https://datasheets.maximintegrated.com/en/ds/MAX30100.pdf

I really wonder who designed this breakout board and why I'm still trying to cover up the worst breakout design I've ever seen (sorry for the rant).

Another thing to try is to use the provided grapher, running the debug example. It's quite insightful and you could eventually paste here annotated graphs of your tests.

Besides, consider that MAX30100 is out of production and if you're serious about integrating a HRS then have a look at MAX30101. If you're still interested in MAX30100 in order to use this library and for didactic reasons, probably you can still find this breakout: https://www.mikroe.com/heart-rate-click (I'm not affiliated)

patty-victor, 2019-03-10:
I’ll try these tipos today evening and I’ll give you a follow up soon.

Thank you!

Outlook für iOS<https://aka.ms/o0ukef> beziehen

patty-victor, 2019-03-11:
Hi @oxullo 

The warm was both with 3V3 or 5V supply. Now I removed IRD and RD connection to ground and the sensor is not getting warm anymore. VICTORY! However, the output is the same. Finger still, no readings. Finger moving (clapping, moving sideways), some weird readings.

I went checking all the connections you suggested and all of them went good. However, when I disconnected the SDA and SCL pins from UNO and let them floating WITH the pullups, I've got 5V on them, which is expected right? If I connected them on ground, then I'd have 0V.

The continuity was also ok in all points you suggested. I didn't run yet the codes you suggested, but I'll soon. In any case, the warming problem is resolved, but the awkward readings not yet. I'll try some other things, then if nothing goes right, I'll change my breakout board. Unfortunally, they are expensive for me :(, so I'll try some other stuff on this one before moving to another.

Thank you!

akshay-solanke, 2020-08-25:
Hi @patty-victor 

I was getting the same issue with my MAX30100 module. Actually the reason for your warming sensor problem is because of the wrong connection. Because after noticing to @oxullo description I compared my connection and I found my mistake.
I was connecting 3.3v to Vin and the result was below after running Tester code on nano  Arduino board:
------------------------------------------------------------
12:03:55.750 -> Initializing MAX30100..Success
12:03:55.750 -> Enabling HR/SPO2 mode..done.
12:03:55.750 -> Configuring LEDs biases to 50mA..done.
12:03:57.471 -> Initializing MAX30100..Success
12:03:57.471 -> Enabling HR/SPO2 mode..done.
12:03:57.471 -> Configuring LEDs biases to 50mA..done.
12:03:58.453 -> Lowering the current to 7.6mA..done.
12:03:59.467 -> Shutting down..done.
12:04:00.460 -> Resuming normal operation..done.
12:04:00.979 -> Sampling die temperature..done, temp=0.00C
12:04:00.979 -> WARNING: Temperature probe reported an odd value
12:04:00.979 -> 
12:04:00.979 -> Press any key to go into sampling loop mode
--------------------------------------------------------------------------
as I was connected 3.3V to Vin and also providing 3.3v Pull up to SDA and SCL with 4.7k the above error was found (temp=0.00C).

Then I switched to Vin from 3.3v to 5v as @oxullo described and now, here is my reasult-

12:04:40.517 -> Initializing MAX30100..Success
12:04:40.517 -> Enabling HR/SPO2 mode..done.
12:04:40.517 -> Configuring LEDs biases to 50mA..done.
12:04:41.504 -> Lowering the current to 7.6mA..done.
12:04:42.518 -> Shutting down..done.
12:04:43.500 -> Resuming normal operation..done.
12:04:44.006 -> Sampling die temperature..done, `temp=28.50C`
12:04:44.006 -> All test pass.
12:04:44.006 -> 
12:04:44.006 -> Press any key to go into sampling loop mode
 Thanks

vaishnavi160, 2021-01-21:
my sensor is getting very hot even if i try it with 5v or 3.3 v. i also tried with 4.7k resistors on sda and scl but still no improvement. i have  even tried testing the code of tester but it isnt working. if anyone knows the solution please help. thanks

vaishnavi160, 2021-01-21:
hello @patty-victor 
i had same issue of sensor getting hot. i tried with both 3.3 and 5 v . also my ird and rd are not connected to ground. even then i m facing the issue. can u help?

akshay-solanke, 2021-01-21:
> hello @patty-victor
> i had same issue of sensor getting hot. i tried with both 3.3 and 5 v . also my ird and rd are not connected to ground. even then i m facing the issue. can u help?

First, tell us detail about your circuit and which type of sensor exactly you are using... mostly in market lots of fake sensor were present..

vaishnavi160, 2021-01-21:
i m using MAX30100 sensor from sunrobotics . i m connecting the circuit:-  vin to 5v,  sda scl to 5 v with 4.7k resistors and int ird and rd are not connected at all.

vaishnavi160, 2021-01-21:
i connected 1.8k resistor to vin and then to 5 v . now its not heating but its not working too.

akshay-solanke, 2021-01-21:
so here is the problem with your sensor module,
1) You are providing a 5v Pullup to SDA and SCL pins with a 4k7 resistor which is not suitable for our sensor circuit.
so, first what you need to change in your circuit is -"just replace Pullup from 5v to 3.3v one ."

another error with the sensor is designed for a 1.8v controller and if you are using something like (Arduino, ARM) with 5v then you should change the sensor circuit itself as shown below:

[image]
(Add jumper like yellow line shown in pic and also cut out the cross marked connection from your sensor as)

Thanks

vaishnavi160, 2021-01-21:
when i changed the pull up from 5 to 3.3 the sensor doesnt work. the light glowing goes dim. and my sensor looks like this
[image]
 i think it has the yellow connection.
thankyou so much.

vaishnavi160, 2021-01-21:
and whenever i m trying to run the code it say intialize success but doesnt go ahead. it never enters the loop i guess. no readings are detected. not even 0

akshay-solanke, 2021-01-21:
can You try with directly connecting 3.3v to Vin of max30100?

vaishnavi160, 2021-01-21:
hello @akshay-solanke 
my sensor doesn't work i guess with 3.3 v cause when i give 5v it starts blinking(acts like working) but with 3.3 it just lights up. and is there any other code for this sensor?

vaishnavi160, 2021-01-22:
12:51:59.470 -> Initializing MAX30100..Success
12:51:59.470 -> Enabling HR/SPO2 mode..done.
12:51:59.470 -> Configuring LEDs biases to 50mA..done.
12:51:59.470 -> Initializing MAX30100..Success
12:51:59.470 -> Enabling HR/SPO2 mode..done.
12:51:59.470 -> Configuring LEDs biases to 50mA..done.
12:52:00.485 -> Lowering the current to 7.6mA..done.
12:52:01.475 -> Shutting down..done.
12:52:02.458 -> Resuming normal operation..done.
12:52:02.970 -> Sampling die temperature..done, temp=127.94C
12:52:02.970 -> All test pass.
12:52:02.970 -> 
12:52:02.970 -> Press any key to go into sampling loop mode

this is my tester code output
is this temperature range normal?

yashrastogi7, 2021-04-26:
@vaishnavi160 did you get your problem resolved? I have the same sensor as you have!

vaishnavi160, 2021-04-26:
> @vaishnavi160 did you get your problem resolved? I have the same sensor as you have!

my sensor was getting heated way too much. what exactly is your sensor problem?

yashrastogi7, 2021-04-26:
@vaishnavi160 i am using nodemcu esp8266 microcontroller, but supplying 5V through external battery to Vin pin. Running tester code it says I2C error(i have checked my connections it's all correct ) and on running MAX30100_PulseOximeter.h it says failed to initialize HRM sensor

vaishnavi160, 2021-04-26:
@yashrastogi7 
1st use pullup and give 3.3 V only. And try to check the circuit again because some pins need to leave open (not connected to anything) IRD RD. and if nothing works, just change the sensor :)

BloggingKIng, 2023-06-23:
Hi there, Did anyone of you get the problem solved? I mean that random data problem. The max30100 showing random data even when the finger is not placed!