oxullo/Arduino-MAX30100 #60: @oxullo

celciusjj, 2019-05-04:
@oxullo 
Connections :
Vin => arduino 3. 3v 
Scl => ardunio Scl 
Sda=>ardunio sda
Int=> ardunio pin 2 
Gnd=> arduino gnd

_Originally posted by @mesutsaygioglu in https://github.com/oxullo/Arduino-MAX30100/issues/13#issuecomment-289034931_

Can you help, i dont know connect this sensor, i put the connections but the led not put on. Thank you

FelipeMdeO, 2019-05-04:
> in #13 (comment)

To have light in led is necessary configure it to do.
Read max30100 datasheet and you will understand better.

celciusjj, 2019-05-04:
yes, I know. But max30100 not working, any peoples say that i need  desolder the resistences, and put 4.7kOhms. What do you know?

sorry for my english, I speak spanish

FelipeMdeO, 2019-05-05:
Sorry for my delay and no problem for your english, I no speak english well yet, I speak portuguese...

So... about change the resistor, wait to do it, this will be last thing to do.

Can you communicate with sensor ? 
You can use library i2c scanner to do it. -  https://playground.arduino.cc/Main/I2cScanner/
Other method to test comunnication is **uncomment** line 18 of sketch: 
`//pulseOxymeter->printRegisters();` and use serial monitor.
If it work:
Try comment line :
`pulseOxymeter = new MAX30100();`
and uncomment line:
`//pulseOxymeter = new MAX30100( DEFAULT_OPERATING_MODE, DEFAULT_SAMPLING_RATE, DEFAULT_LED_PULSE_WIDTH, DEFAULT_IR_LED_CURRENT, true, true );`

Post your result, please.

celciusjj, 2019-05-05:
Ok I try that. Dont worry, thank you for help me.

I have two questions.
1-  what scheme did you use to mount the circuit

2- Did it work with the soldered resistors of the max30100?

tomorrow I try the communication l2c, today i am a little busy.

FelipeMdeO, 2019-05-05:
**1- what scheme did you use to mount the circuit:**
I use standard connection. At this time I´m not using Arduino, but NXP Freedom kl02 board. I only connect Vcc, GND, SDA and SCL and it worked.
**2- Did it work with the soldered resistors of the max30100?**
I´m using this board: -> 
https://pt.aliexpress.com/item/WS16-NEW-MAX30100-Heart-Rate-Sensor-Click-Heart-Rate-Oximeter-Pulse-Sensor-Pulsesensor-Module-For-Arduino/32722466324.html?spm=a2g0s.9042311.0.0.556cb90aJzwqFv
For it I no need do any modification in board.

celciusjj, 2019-05-05:
Oh, ok i am use the max30100 green or rcwl - 0530, this is the best tutorial that i found.
https://www.teachmemicro.com/max30100-arduino-heart-rate-sensor/

FelipeMdeO, 2019-05-05:
uhnnnnnn...
Basead in your tutorial, you will really need remove pull ups connecteds to SDA and SCL and use resistor to +5V to do it work in Arduino.
At my case, maybe work because my micro controller is +1.8 to 3.6 V.
So, do it. Remove pull ups and use external resistor connected between +5V and SLC and +5V and SDA and try use Arduino library.

I wait for news!
Good luck.

celciusjj, 2019-05-07:
Hello again, the led now is on, but in the serial 
Initializing pulse oximeter..SUCCE%
Heart rate:53.05bpm / SpO2:1%
68.69bpm / SpO2:0%
Beat!
Heart rate:109.85bpm / SpO2:1%
Beat!
Heart rate:53.05bpm / SpO2:1%
Heart rate:53.05bpm / SpO2:1%
⸮q⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮(⸮⸮ْ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮Yc⸮⸮.⸮Yk⸮[k⸮Y

⸮⸮⸮⸮⸮)⸮⸮⸮9⸮⸮c⸮⸮⸮⸮⸮⸮i⸮⸮⸮⸮⸮⸮)⸮⸮⸮⸮⸮+⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮)⸮⸮⸮⸮⸮5⸮⸮Yc⸮[c⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ᘉ⸮⸮)⸮⸮

Any times the led is on, but other times is off.

FelipeMdeO, 2019-05-07:
What did you do?
Change you pull ups ?

FelipeMdeO, 2019-05-07:
Verify your baud rate of serial.

celciusjj, 2019-05-07:
I have 115200, what value i put?

FelipeMdeO, 2019-05-07:
You need put same value in serial monitor. Do you verify it?

celciusjj, 2019-05-07:
Yeah i have this value, now if i put RD to ground the led start glow but in program is failed. Is so rare... Sometimes the LED is bright, others do not shine or shine little.

in this post https://github.com/oxullo/Arduino-MAX30100/issues/16
saikapilan have the same problem.

FelipeMdeO, 2019-05-07:
You tried connecting just VCC, GND, SDA and SCL?
I just use that pins.
And... you put pull ups?

celciusjj, 2019-05-07:
Yeah, i am use this pins, but the led not put on, when i put RD to ground the led is glow, also when i put my finger above him.

Sometimes, when connect arduino the led put on, but inmediatly this put off. Jajaja omg

celciusjj, 2019-05-07:
[https://photos.google.com/search/_tra_/photo/AF1QipN0AUEq0QFGmCam9G10D-8YCy23_zRGq-X3yIRI](url)
two resistences vin to 5v
scl to scl arduino,
sda to sda arduino
and ground to ground arduino

celciusjj, 2019-05-07:
Forget the previous comments, I turn on. Now let's look at the data that the sensor delivers.

https://photos.google.com/search/_tra_/photo/AF1QipPWkfs8XfOd0t84Jjk6XW7lupYK7-PHM3vb5eMW

look this, I put the finger to take the pines with the sensor.

bibhutibikramaditya, 2020-07-30:
searching for source code that can run and give SPO2 level even less than 85 in case of covid 19 patient. SPO2 LEVEL should be anything from 0 to 100