oxullo/Arduino-MAX30100 #63: no red light Arduino Uno + MAX30100

gag2012, 2019-05-27:
### Troubleshooting checklist

- [ check ] I read the README (on master) thoroughly
- [ check ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ check ] I filled in all the details of my setup down below

### Description of the issue

### Output from MAX30100_Tester example

### Details of my setup

* Arduino hardware: Arduino UNO
* MAX30100 breakout: Green Board
* Arduino framework version: Arduino 1.8.5
* MAX30100 library version: 1.2.1

my connection :

arduino UNO      <-------->       MAX30100
       3.3v                                            Vin
       A5                                              SCL           (4.7kohm pull up)
       A4                                              SDA          (4.7kohm pull up)       
       D2                                              INT          (4.7kohm pull up)
      GND                                          GND

the code is MAX30100_Minimal

[image]

and I removed three register solder and connect external 4.7k ohm register.  

3days ago my MAX30100 sensor work well like below picture.

[image]

but now it doesn't work same in spite of the same circumstances.

I don't know why my MAX30100 sensor red light isn't on.

Apparently, it worked well until three days ago.

[image]

I read all past issue and try that (change register, bread board, convert 3.3v to 5v) but doesn't work :(

[image]

and When it is 3.3v, it outputs 'Initializing pulse oximeter..FAILED' and when it is 5v it prints 'Initializing pulse oximeter..'.

and when it is 5v the MAX30100 sensor red light comes in for about half a second very short.

Is there anyone who can help me?

sorry for my english.

zelalem2, 2019-06-01:
Can some one comment on this, the same thing happend to me too.

lolybob4345, 2019-07-25:
This may help you: https://reedpaper.wordpress.com/2018/08/22/pulse-oximeter-max30100-max30102-how-to-fix-wrong-board/

allenchak, 2019-11-25:
I have same issue, but after some google and tries. Its work fine with **ProMini 3.3v** and without changing the PCB.