oxullo/Arduino-MAX30100 #64: max30100 with esp8266 cant send the value to web

Hui0924, 2019-08-01:
I used two esp8266 one for server one for client , I try to send the value of Max30100 but it always 0
[新文字文件 (2).txt](https://github.com/oxullo/Arduino-MAX30100/files/3455206/2.txt)

happynet95, 2019-09-27:
Did you make some progress on this issue? I'm suffering from the same problem. ToT

kentpaul, 2020-04-26:
hello sir can i see the circuit diagram and code? im having an issue with my max30100 and nodemcu esp8266, when i open my serial monitor i only get soft wdt reset. i need help sir please

`Soft WDT reset

stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01b0
3fffff90: feefeffe feefeffe feefeffe 3ffee650
3fffffa0: 3fffdad0 00000000 3ffee610 402025cc
3fffffb0: feefeffe feefeffe 3ffe84ec 40100ea1
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Initializing MAX30100..FAILED: I2C error`

happynet95, 2020-04-28:
> hello sir can i see the circuit diagram and code? im having an issue with my max30100 and nodemcu esp8266, when i open my serial monitor i only get soft wdt reset. i need help sir please
>
> `Soft WDT [...]
`
void loop() {

  if(index_minute%5==0){
    pox.begin();
  }
  if(power){
    UpdateSensors();
    SNB_task();
    send_data();
  }`
Hi. I solved this problem in the end, In my situation, it turned out that the pox sensor stops frequently when I use bluetooth or Wi-Fi. So I just put some code that makes MAX30100 to restart periodically.

here is my code:

Moreover, there were some issues with setup function. I don't know why, but the order between Bluetooth setup, pox setup, etc matters. Try changing their orders, or inserting short delays between codes. here is my code:

`
void SensorSetup() {
  
  // initialize both serial ports:
  //Serial.begin(115200);
  
  Wire.begin();
  
  // Temperature Sensor initializing
   //// spo2 Sensor initializing ///
  
  mpu.setup();
  
  delay(2000);
  
  SerialBT.begin(device_key);
  
  //Serial.print("Initializing PulseOximeter..");
  if (!pox.begin()) {
        //Serial.println("FAILED");
  } else {
        //Serial.println("SUCCESS");
  }
}
`

kentpaul, 2020-04-28:
thanks sir big help, btw can i copy your whole code? im very new to this sir

kentpaul, 2020-04-29:
Btw sir did you also use a temperature sensor in your project? that temperature sensor initializing.

kentpaul, 2020-05-11:
hello sir what is mpu? i need help please

kentpaul, 2020-05-30:
Hello sir did you solve your problem? Can you now send to web? Im having the same issue as yours