oxullo/Arduino-MAX30100 #66: What is the range of SpO2 readings we can get after using this library ?

ktgrhsd, 2019-09-19:
Is 93% the lowest it can go ? 

I was able to run MAX30100 sensor with Arduino Uno properly. Verified my readings with a medical grade device and got fairly accurate readings, hence not using the given issue template.

FarzanehK, 2020-07-27:
> Is 93% the lowest it can go ? Where and what to change if we want to detect lower values as well ?

Dear Ktgrhsd

I have the same problem with MAX30100 (green version) that SpO2% does not reduce to less than 93% even for hypoxia patients. Have you solved your problem? Would you please kindly advice me?

dgok50, 2020-10-12:
I came across the same problem and tried to find its source, as a result I found the following code in MAX30100_SpO2Calculator.c:
`
const uint8_t SpO2Calculator::spO2LUT[43] = {100,100,100,100,99,99,99,99,99,99,98,98,98,98,
                                             98,97,97,97,97,97,97,96,96,96,96,96,96,95,95,
                                             95,95,95,95,94,94,94,94,94,93,93,93,93,93};
`

This means that this library, in principle, is not capable of producing a result less than 93%.
Moreover, due to the lack of index checking when accessing array elements, it is possible to go beyond the array boundaries, which in turn can lead to the output of random data.
`        if (beatsDetectedNum == CALCULATE_EVERY_N_BEATS) {
            float acSqRatio = 100.0 * log(redACValueSqSum/samplesRecorded) / log(irACValueSqSum/samplesRecorded);
            uint8_t index = 0;

            if (acSqRatio > 66) {
                index = (uint8_t)acSqRatio - 66;
            } else if (acSqRatio > 50) {
                index = (uint8_t)acSqRatio - 50;
            }
            reset();

            spO2 = spO2LUT[index];
        }`

I would like the developer to either indicate in the description that the library cannot calculate less than 93, or add the rest of the data to the array.