oxullo/Arduino-MAX30100 #69: max30100 get zero value when i connect to esp8266

0919834492, 2020-02-16:
### Troubleshooting checklist

- [ ] I read the README (on master) thoroughly
- [ ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ ] I filled in all the details of my setup down below

### Description of the issue
[image]

After take out the SDA,SCL cable and take it in

[image]

Without "take out and in SDA.SCL cable", we are unable to get heartrate value. it stuck like pic 1.
### Output from MAX30100_Tester example
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=29.19C
All test pass.

Press any key to go into sampling loop mode
IR=6679 RED=10509
IR=6676 RED=10504
IR=6675 RED=10509
IR=6668 RED=10510
IR=6673 RED=10515
IR=6675 RED=10507
IR=6682 RED=10508
IR=6678 RED=10509
IR=6679 RED=10516
IR=6683 RED=10517
IR=6687 RED=10514
IR=6688 RED=10517
IR=6683 RED=10511
IR=6683 RED=10511
IR=6676 RED=10500
### Details of my setup

* Arduino hardware: NodeMCU ESP8266
* MAX30100 breakout: GY-Max30100
* Arduino framework version: 1.8.12
* MAX30100 library version: 1.2.1

kentpaul, 2020-04-26:
hello sir can i see the circuit diagram and code? im having an issue with my max30100 and nodemcu esp8266, when i open my serial monitor i only get soft wdt reset. i need help sir please

`Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01b0
3fffff90:  feefeffe feefeffe feefeffe 3ffee650  
3fffffa0:  3fffdad0 00000000 3ffee610 402025cc  
3fffffb0:  feefeffe feefeffe 3ffe84ec 40100ea1  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Initializing MAX30100..FAILED: I2C error
`

kentpaul, 2020-04-27:
can i see the circuit diagram sir?

kentpaul, 2020-04-28:
did you solve this problem sir? please i need help thank ou

yimihua88, 2020-04-30:
some things take time