oxullo/Arduino-MAX30100 #70: Big problem - cannot use this library in any serious project (big pulse delta or showing pulse without finger)

DarkDaemon1, 2020-04-05:
### Troubleshooting checklist

- [x] I read the README (on master) thoroughly
- [x] I ran the MAX30100_Tester and I'm going to paste the output down below
- [x] I filled in all the details of my setup down below

### Description of the issue
When I add delay more than 30ms in loop with pox.update(), I get wrong pulse data with big delta.  When I add delay more than 80ms in dump function with 1s perion, I get phantom pulse data without finger. So, it is not possible to use this library in any serious project with other functions, which take a lot of processor time. Anyone know the solution?

> Heart rate:53.04bpm / SpO2:93%
Beat!
Heart rate:56.98bpm / SpO2:93%
Beat!
Beat!
Heart rate:83.81bpm / SpO2:93%
Beat!
Heart rate:95.21bpm / SpO2:93%
Beat!
Beat!
Heart rate:67.57bpm / SpO2:93%
Beat!
Heart rate:65.02bpm / SpO2:93%
Heart rate:65.02bpm / SpO2:93%
Beat!
Heart rate:57.93bpm / SpO2:93%
Beat!
Heart rate:38.50bpm / SpO2:93%
Beat!
Beat!
Heart rate:104.25bpm / SpO2:93%
Beat!
Heart rate:73.49bpm / SpO2:95%
Beat!
Heart rate:64.13bpm / SpO2:95%

### Output from MAX30100_Tester example

> Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=40.25C
All test pass.
Press any key to go into sampling loop mode

### Details of my setup

* Arduino hardware: Arduino Pro mini 16MHz 5V
* MAX30100 breakout: RCWL-0530
* Arduino framework version: 1.8.12
* MAX30100 library version: 1.2.1

DarkDaemon1, 2020-04-05:
I found solution, that increase time for other functions twice. I slow down sample rate to 50Hz.

I've added in MAX30100_PulseOximeter.h
`void setSamplingRate(SamplingRate sr);`

in MAX30100_PulseOximeter.cpp
```
void PulseOximeter::setSamplingRate(SamplingRate sr)
{
    hrm.setSamplingRate(sr);
}
```
And then use
`pox.setSamplingRate(MAX30100_SAMPRATE_50HZ);`

It is not perfect solution, but it fits to my project

chandravegeta, 2020-08-13:
I have also seen this issue, its due to light from tubelight or CFL or LED light falling into Senosr.