oxullo/Arduino-MAX30100 #72: Max30100/Max30102 the led is not lit up and i get a soft wdt reset

kentpaul, 2020-04-27:
### Troubleshooting checklist

- [ ] I read the README (on master) thoroughly
- [ ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ ] I filled in all the details of my setup down below

### Description of the issue
i put 4.7k pull ups but still the same i only get this error. And without the pull ups still i only get this error. And lastly when i unplug my pins to nodemcu then i run upload, serial monitor im still getting this error
[image]

### Output from MAX30100_Tester example
[image]

### Details of my setup

* Arduino hardware: NODEMCU ESP8266
* MAX30100 breakout: RCWL-0531 Max301002 the same design of max30100
* Arduino framework version: 1.8.12
* MAX30100 library version: 1.2.1

cerebrum2020, 2020-04-28:
Hello, 

maybe try with i2c scanner

https://playground.arduino.cc/Main/I2cScanner/

Also on my configuration if I connect just the MAX30100 board (standard, without mod) no way to work. If I connect also the i2c dispaly ot work correctly.

kentpaul, 2020-04-29:
hello sir this is the ouput that i get
Scanning...
Unknown error at address 0x3A
I2C device found at address 0x57  !
done

Scanning...
Unknown error at address 0x0D
I2C device found at address 0x57  !
I2C device found at address 0x5A  !
done

Scanning...
I2C device found at address 0x57  !
done

Scanning...
I2C device found at address 0x57  !
Unknown error at address 0x78
done

Scanning...
I2C device found at address 0x31  !
I2C device found at address 0x57  !
done

Scanning...
Unknown error at address 0x1C
I2C device found at address 0x20  !
I2C device found at address 0x57  !
done

Scanning...
I2C device found at address 0x57  !
I2C device found at address 0x58  !
done

kentpaul, 2020-05-03:
whats next sir? i dont know how to disable the wtd reset

cerebrum2020, 2020-05-04:
Hello. but is a continuos reading or the different readings are during different tests?

closed 2020-05-28 by kentpaul