oxullo/Arduino-MAX30100 #73: i get zero values when i send the data to database

kentpaul, 2020-05-28:
### Troubleshooting checklist

- [# ] I read the README (on master) thoroughly
- [ #] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ #] I filled in all the details of my setup down below

### Description of the issue
i want to send data to database but i get zero values when i put the Sending_To_phpmyadmindatabase function. The sensor works fine in the example code.

heres my code:

#include <ESP8266mDNS.h>
#include <ESP8266mDNS_Legacy.h>
#include <LEAmDNS.h>
#include <LEAmDNS_lwIPdefs.h>
#include <LEAmDNS_Priv.h>

#include <ESP8266WiFi.h>
#include <DallasTemperature.h>
#include <OneWire.h>
#include <ESP8266HTTPClient.h>

//#include <WiFi.h>    
#include <SPI.h>
#include <Ethernet.h>

#include <Wire.h>
#include "MAX30100_PulseOximeter.h"

#define REPORTING_PERIOD_MS     1000
float BPM, SpO2;
PulseOximeter pox;
uint32_t tsLastReport = 0;

IPAddress server_addr(192,168,15,4);
const char* ssid="pldt";
const char* password = "katrinapangit";
const char *host = "192.168.15.4";

WiFiClient client;  

// Callback (registered below) fired when a pulse is detected
void onBeatDetected()
{
    Serial.println("Beat!");
}

void setup()
{
    Serial.begin(115200);
    Serial.println();
    Serial.print("Wifi connecting to ");
    Serial.println( ssid );

    WiFi.begin(ssid,password);

    while( WiFi.status() != WL_CONNECTED ){
        delay(1000);
        Serial.print(".");    

   }
    Serial.println();
    Serial.print("wifi connected");

    Serial.println("Server started");
    Serial.println(WiFi.localIP() );
    delay(1000);
    Serial.println("connecting......");

    Serial.print("Initializing pulse oximeter..");
    if (!pox.begin()) {
        Serial.println("FAILED");
        for(;;);
    } else {
        Serial.println("SUCCESS");
    }
    pox.setIRLedCurrent(MAX30100_LED_CURR_7_6MA);

    pox.setOnBeatDetectedCallback(onBeatDetected);
}

void loop()
{
  pox.update();
  BPM = pox.getHeartRate();
  SpO2 = pox.getSpO2();
  
  
  if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
    Serial.print("BPM: ");
    Serial.println(BPM);
    Serial.print("SpO2: ");
    Serial.print(SpO2);
    Serial.println("%");
    Serial.println("*********************************");
    Serial.println();
    tsLastReport = millis();
    Sending_To_phpmyadmindatabase(BPM);
    }
}
void Sending_To_phpmyadmindatabase(float BPM){

  if (client.connect(server_addr,80)) {
    Serial.println("connected");
    Serial.print("GET /pulse.php?BPM=12");
    client.print("GET /pulse.php?BPM="+String(BPM,1));
    Serial.println("BPM!");
    client.println("pulse="); 
    client.println(BPM);
   // Serial.print("GET /vsmonitor/connect.php?SpO2=12");
   // client.print("GET /vsmonitor/connect.php?SpO2="+String(SpO2,1));
   // Serial.println("SpO2!");
    //client.println("o2_sat="); 
    //client.println(SpO2);
    client.print(" ");
    client.println(" HTTP/1.1"); 
    client.println("Host: 192.168.15.4");
    client.println(server_addr);
    client.println("Connection: close"); 
    client.println(); 
  }
  else {
    Serial.println("--> connection failed\n");
  }
  Serial.println();
  delay(5000);

}

### Output from MAX30100_Tester example

### Details of my setup

* Arduino hardware: nodemcu esp8266
* MAX30100 breakout: Max30100
* Arduino framework version: 1.8.12
* MAX30100 library version: latest version

MIIB1, 2020-12-20:
I have similar problem when i send data to thingspeak it give zero values, i don't know why ?
did you fix the problem ?

gatorkoco, 2021-01-30:
what about your php code ? 
i have same issue with you but I don't know how to implement HTTP post in my code ??
can u show me ur PHP code ?

ZahraaSabah97, 2021-05-19:
> I have similar problem when i send data to thingspeak it give zero values, i don't know why ?
> did you fix the problem ?

currently, I'm having exactly the same problem, if you fixed the problem please tell me how...

MIIB1, 2021-05-20:
> > I have similar problem when i send data to thingspeak it give zero values, i don't know why ?
> > did you fix the problem ?
> 
> currently, I'm having exactly the same problem, if you fixed the problem please tell me how...

i don't remember now l need to access my file but i think the problem was with the delay.
you can post you code so we can help you

ZahraaSabah97, 2021-05-20:
> i don't remember now l need to access my file but i think the problem was with the delay.
> you can post you code so we can help you

Here is my code:

```
#define USE_ARDUINO_INTERRUPTS true    
#define DEBUG true
#define SSID "******"           // "SSID-WiFiname"
#define PASS "******"      // "password"
#define IP "184.106.153.149"      // thingspeak.com ip
#define REPORTING_PERIOD_MS     1000
#include <Event.h>
#include <Timer.h>
#include <Wire.h>
#include "MAX30100_PulseOximeter.h"
#include <SoftwareSerial.h>
#include "Timer.h"
Timer t;

PulseOximeter pox;
uint32_t tsLastReport = 0;

String msg = "GET /update?key=ETIEQG6HXQYK2TWG";    //API key
SoftwareSerial esp8266(2,3);    //RX, TX

//Variables
float myTemp;
int myBPM;
int mySpO2;
String BPM;
String temp;
int error;
int raw_myTemp;
float Voltage;
float tempC;

void onBeatDetected()
{
    Serial.println("Beat!");
}

void setup()
{
    Serial.begin(9600); 
    esp8266.begin(115200);
    Serial.print("Initializing pulse oximeter..");

    if (!pox.begin()) {
        Serial.println("FAILED");
        for(;;);
    } else {
        Serial.println("SUCCESS");
    }
    pox.setOnBeatDetectedCallback(onBeatDetected);
    
    Serial.println("AT");
    esp8266.println("AT");
    delay(3000);        
      
    if(esp8266.find("OK"))
    {
      connectWiFi();
    }
     t.every(10000, getReadings);
     t.every(10000, updateInfo);
}

void loop()
{
   pox.update();
   myBPM = pox.getHeartRate();
   mySpO2 = pox.getSpO2();
    if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
        Serial.print("Heart rate: ");
        Serial.print(myBPM);
        Serial.print("bpm / SpO2: ");
        Serial.print(mySpO2);
        Serial.println("%");

        tsLastReport = millis();
}
 start: //label
 error=0;
   t.update();    //Resend if transmission is not completed
 if (error==1)
  {
  goto start;     //go to label "start"
  } 
  delay(4000);      
}

void updateInfo()
{
  String cmd = "AT+CIPSTART=\"TCP\",\"";
  cmd += IP;
  cmd += "\",80";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(2000);
 
  if(esp8266.find("Error"))
  {
    return;
  }
  cmd = msg ;
  cmd += "&field1=";    //field 1 for BPM
  cmd += BPM;
  cmd += "&field2=";  //field 2 for temperature
  cmd += temp;
  cmd += "\r\n";
  Serial.print("AT+CIPSEND=");
  esp8266.print("AT+CIPSEND=");
  Serial.println(cmd.length());
  esp8266.println(cmd.length());
 
  if(esp8266.find(">"))
  {
    Serial.print(cmd);
    esp8266.print(cmd);
  }
  else
  {
    Serial.println("AT+CIPCLOSE");
    esp8266.println("AT+CIPCLOSE");
    //Resend...
    error=1;
  }
}

boolean connectWiFi()
{
  Serial.println("AT+CWMODE=1");
  esp8266.println("AT+CWMODE=1");
  delay(2000);
  String cmd="AT+CWJAP=\"";
  cmd+=SSID;
  cmd+="\",\"";
  cmd+=PASS;
  cmd+="\"";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(5000);
  
  if(esp8266.find("OK"))
  {
    return true;
  }
  else
  {
    return false;
  }
}

void getReadings()
{
  raw_myTemp = analogRead(A1);
  Voltage = (raw_myTemp / 1023.0) * 5000;       // 5000 to get millivots.
  tempC = Voltage * 0.1; 
  myTemp = (tempC * 1.8) + 32;                  // conver to F
  Serial.print("TEMP= ");
  Serial.println(myTemp);
  myBPM= pox.getHeartRate();
    delay(20);            
    char buffer1[10];
    char buffer2[10];
    BPM  = dtostrf(myBPM, 4, 1, buffer1);
    temp = dtostrf(myTemp, 4, 1, buffer2);  
  }
```

ZahraaSabah97, 2021-05-21:
> i don't remember now l need to access my file but i think the problem was with the delay.
> you can post you code so we can help you

I hope you can help me...

MIIB1, 2021-05-22:
> I hope you can help me...

Hi
sorry for late replay i was busy, did you try to put
#include "ThingSpeak.h"
#define SECRET_CH_ID xxxxxxxx      // replace 0000000 with your channel number
#define SECRET_WRITE_APIKEY "xxxxxxxxxx"   // replace XYZ with your channel write API Key

i don't see it in your code ?

gatorkoco, 2021-05-22:
I can post to database(mysql) but .. here comes the other problem .. after several data posted to database .. the reading from max30100 become less accurate .. and sometimes it post the same value .. 

I use max30100 and nodemcu ..

MIIB1, 2021-05-22:
> I can post to database(mysql) but .. here comes the other problem .. after several data posted to database .. the reading from max30100 become less accurate .. and sometimes it post the same value ..
> 
> I use max30100 and nodemcu ..

Hi
That normal because max30100 not high quality it for testing  you can store the value like 10 value after that send it to database.

MohamedZaky0, 2021-06-21:
> Hi
> That normal because max30100 not high quality it for testing you can store the value like 10 value after that send it to database.

Hey man sorry for the inconvenience But i am trying to use AD8232 with Max30100 , the Max30100 had some hardware modification and i did it and i get right values with it alone with Nodemcu V2 / Arduino Uno . now if i use Nodemcu and try to upload the values to Ubidots when i merge the code of Max30100 with uploading code the values are 0s and if i added just the Max30100 with code with another sensor AD8232 it also get 0 or just restart and call void Setup more and more .. i have been in this issue for days dunno what to do

gatorkoco, 2021-06-24:
> after that send it to database.

aaaaahh i see... thx for ur help sir..

Bryanhoody, 2021-09-15:
> Saya dapat memposting ke database (mysql) tetapi .. ini dia masalah lain .. setelah beberapa data diposting ke database .. pembacaan dari max30100 menjadi kurang akurat .. dan kadang-kadang memposting nilai yang sama ..
> 
> Saya menggunakan max30100 dan nodemcu ..

Bryanhoody, 2021-09-15:
Hy can i ask for the whole source code, i hope you can help me because for my final project

gatorkoco, 2021-09-16:
> Hy can i ask for the whole source code, i hope you can help me because for my final project

hi ..
i'm not sure about my code .. after several test it post same value over and over again still trying to fix that problem tho.

Ahmedhatem1234, 2023-04-13:
> ### Troubleshooting checklist
> * [# ] I read the README (on master) thoroughly
> * [ #] I ran the MAX30100_Tester and I'm going to paste the output down below
> * [ #] I filled in all the details of my [...]

Ahmedhatem1234, 2023-04-13:
have you found the solution to this problem because i face the same problem?