oxullo/Arduino-MAX30100 #74: Unable to Have HTTP Post request to my server using Max30100

sajaldegal, 2020-06-03:
I am using MAX30100  to post BPM and SpO2 to my server using Http Post request and Node MCU the code is compiling with no errors but Iam getting 0.00 value or posting only one value all the time and value not getting updated to my server if I comment the nt httpCode = http.POST(postData); then the sensor is providing the updated values of BPM and SpO2. Please help me in this issue. Iam attaching my file below.

 #include "MAX30100_PulseOximeter.h"
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>

#include <Wire.h>
#include "Adafruit_GFX.h"
#include "OakOLED.h"

#define REPORTING_PERIOD_MS 1000
OakOLED oled;
/* Set these to your desired credentials. */
const char *ssid = "Nokia 5";  //ENTER YOUR WIFI SETTINGS
const char *password = "priyanka@333";
 
//Web/Server address to read/write from 
const char *host = "iot.circuitadda.in";   //https://circuits4you.com website or IP address of server

// Connections : SCL PIN - D1 , SDA PIN - D2 , INT PIN - D0
PulseOximeter pox;

float BPM, SpO2;
float hsa, ssa;
char buffer[10];
char buffer2[10];
String ADCData, ADCData1, postData;
uint32_t tsLastReport = 0;
volatile boolean heartBeatDetected = false;
uint32_t hrt= 0;
#define SLEEP_TIME_SECONDS 1800

 
const unsigned char bitmap [] PROGMEM=
{
0x00, 0x00, 0x00, 0x00, 0x01, 0x80, 0x18, 0x00, 0x0f, 0xe0, 0x7f, 0x00, 0x3f, 0xf9, 0xff, 0xc0,
0x7f, 0xf9, 0xff, 0xc0, 0x7f, 0xff, 0xff, 0xe0, 0x7f, 0xff, 0xff, 0xe0, 0xff, 0xff, 0xff, 0xf0,
0xff, 0xf7, 0xff, 0xf0, 0xff, 0xe7, 0xff, 0xf0, 0xff, 0xe7, 0xff, 0xf0, 0x7f, 0xdb, 0xff, 0xe0,
0x7f, 0x9b, 0xff, 0xe0, 0x00, 0x3b, 0xc0, 0x00, 0x3f, 0xf9, 0x9f, 0xc0, 0x3f, 0xfd, 0xbf, 0xc0,
0x1f, 0xfd, 0xbf, 0x80, 0x0f, 0xfd, 0x7f, 0x00, 0x07, 0xfe, 0x7e, 0x00, 0x03, 0xfe, 0xfc, 0x00,
0x01, 0xff, 0xf8, 0x00, 0x00, 0xff, 0xf0, 0x00, 0x00, 0x7f, 0xe0, 0x00, 0x00, 0x3f, 0xc0, 0x00,
0x00, 0x0f, 0x00, 0x00, 0x00, 0x06, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
 
void onBeatDetected()
{
    heartBeatDetected = true;
    Serial.println("Beat Detected!");
    oled.drawBitmap( 60, 20, bitmap, 28, 28, 1);
    oled.display();
} 
 

 
void setup()
{
  
  Serial.begin(115200);
    oled.begin();
    oled.clearDisplay();
    oled.setTextSize(1);
    oled.setTextColor(1);
    oled.setCursor(0, 0);
 
    oled.println("Initializing pulse oximeter..");
    oled.display();
    
  pinMode(16, OUTPUT);
  WiFi.mode(WIFI_OFF);        //Prevents reconnection issue (taking too long to connect)
  delay(1000);
  WiFi.mode(WIFI_STA);        //This line hides the viewing of ESP as wifi hotspot
  
  WiFi.begin(ssid, password);     //Connect to your WiFi router
  Serial.println("");
 
  Serial.print("Connecting");
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

   Serial.print("Initializing Pulse Oximeter..");
if (!pox.begin())
    {
         Serial.println("FAILED");
         oled.clearDisplay();
         oled.setTextSize(1);
         oled.setTextColor(1);
         oled.setCursor(0, 0);
         oled.println("FAILED");
         oled.display();
         for(;;);
    }
    else
    {
         oled.clearDisplay();
         oled.setTextSize(1);
         oled.setTextColor(1);
         oled.setCursor(0, 0);
         oled.println("SUCCESS");
         oled.display();
         Serial.println("SUCCESS");
         
    }
    pox.setOnBeatDetectedCallback(onBeatDetected);
}
 
void loop()
{
  pox.update();
        BPM = pox.getHeartRate();
    SpO2 = pox.getSpO2();
      
  //Serial.println(ADCData);
  
  //Serial.println(ADCData1);
  
  //Post Data
  
  //int adcvalue=BPM;
  //ADCData = String(adcvalue);   //String to interger conversion
  
  //int adcvalue1=SpO2;
  //ADCData1 = String(adcvalue1);   //String to interger conversion
  
 
  
  if (millis() - tsLastReport > REPORTING_PERIOD_MS)
    {
        Serial.print("Heart rate:");
        Serial.print(pox.getHeartRate());
        Serial.print("SpO2:");
        Serial.print(pox.getSpO2());
        Serial.println(" %");
     
        
        oled.clearDisplay();
        oled.setTextSize(1);
        oled.setTextColor(1);
        oled.setCursor(0,16);
        oled.println(pox.getHeartRate());
 
        oled.setTextSize(1);
        oled.setTextColor(1);
        oled.setCursor(0, 0);
        oled.println("Heart BPM");
 
        oled.setTextSize(1);
        oled.setTextColor(1);
        oled.setCursor(0, 30);
        oled.println("Spo2");
 
        oled.setTextSize(1);
        oled.setTextColor(1);
        oled.setCursor(0,45);
        oled.println(pox.getSpO2());
        oled.display();

        
        tsLastReport = millis();

        
   if (heartBeatDetected && BPM  != 0) {
    
   
    ADCData1= String(BPM, 2);
  //ADCData= dtostrf(SpO2, 2, 2, buffer);
    //ADCData1 = String(pox.getHeartRate(), 2);
    //ADCData1 = "A";
    ADCData = String(SpO2, 2);
    //ADCData = "B";
    postData = "status=" + ADCData + "&status2=" + ADCData1 ;
  HTTPClient http;    //Declare object of class HTTPClient
  http.begin("http://iot.circuitadda.in/post.php");              //Specify request destination
  http.addHeader("Content-Type", "application/x-www-form-urlencoded");    //Specify content-type header
  int httpCode = http.POST(postData);   //Send the request
 Serial.println(httpCode);   //Print HTTP return code
 String payload = http.getString();    //Get the response payload
 Serial.println(payload);    //Print request response payload
 http.end();  //Close connection
 //Serial.print("Test");
   }
    }
}

pahart2208, 2020-06-17:
Hi. Yes, I've struggled with this for a few days. Works fine with the example code 'as-is' when writing to serial or LCD, but when I add anything, the output from MAX30100 stops. In my case I was trying to write to my cloud services dashboard using WiFi.
I'm no s/w expert, but....Seems like the pox.update() has to run at a minimum 100Hz i.e. cycle time of <10ms. I put a couple of software timers in the code and when the example code runs, it takes between 1 and 2 ms, which is easily fast enough. However, the Wifi write was taking around 350ms and so pox.update() stopped working properly.
Now I understand the comment in the code:
// Make sure to call update as fast as possible
At least 100 times each second, or it won't work!
So, your other function must be taking longer than 10ms which stops pox.update() working.
Still trying to figure out a workaround.
Hope that helps - or at least gives some insight into the problem.

Bryanhoody, 2021-09-21:
Apakah bisa anda melampirkan source code yang sudah diperbaiki