oxullo/Arduino-MAX30100 #75: Sensor works fine with simple codes, but not when embedded in a larger project

Ardminu, 2020-07-02:
### Troubleshooting checklist

- [ /] I read the README (on master) thoroughly
- [ /] I ran the MAX30100_Tester and I'm going to paste the output down below
- [/ ] I filled in all the details of my setup down below

### Description of the issue
My MAX30100 sensor (board that contains the pull up resistor) works fine with your library and gives reading with simple codes like MAX30100_Minimal.ino.
However, when I tried to use the sensor in an actual project that contains multiple inputs/outputs, the sensor is not responding properly. When I run the code in debug mode, the sensor get connected successfully, but the sensor.update() did not print any thing on the serial monitor. Sometimes if I disconnect and reconnect the sensor board, 8 pair measurements are printed once on the serial monitor. 
Briefly, my project uses Arduino mega 2560 to read 8 analog signals, 6 digital, and write 2 PWMs. It has also LCD and serial communication. 
The sensor works fine with the LCD, so no issue from the LCD side.
At first, I thought that the lengthy sketch maybe not satisfying the 100Hz sampling rate required by the sensor. I changed the code to update values from the analog and digital pins every half second, and this makes the loop() iteration taking only ~5ms. However, this did not solve the issue. 
I stopped the serial communication and the PWM output, but nothing changed. 

I do not suspect a hardware problem as the sensor works fine with the simple sketch. I do connect only SDA & SCL, but not interrupt which is not utilized by the library as I understand. 

I am assuming a timing or interrupt problem, but could not figure it out. Any help would be appreciated.
 
### Output from MAX30100_Tester example
Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=34.31C
All test pass.

Press any key to go into sampling loop mode
IR=0 RED=15
IR=4 RED=0
IR=8 RED=0
IR=0 RED=0
IR=0 RED=1
IR=0 RED=15
IR=6 RED=0
IR=11 RED=0
IR=0 RED=0
IR=0 RED=0
IR=0 RED=13
IR=1 RED=0
IR=2 RED=0
IR=0 RED=0
IR=0 RED=7
IR=3 RED=6
IR=7 RED=0
IR=9 RED=0
IR=0 RED=0
IR=0 RED=10
IR=0 RED=3
IR=3 RED=0
IR=0 RED=0
IR=0 RED=2
IR=0 RED=12
IR=4 RED=0
IR=7 RED=0
IR=2 RED=0
IR=0 RED=1
IR=0 RED=7
IR=2 RED=0
IR=10 RED=0
IR=0 RED=0
IR=0 RED=5
IR=0 RED=4
IR=10 RED=0
IR=3 RED=0
IR=0 RED=0
IR=0 RED=11
IR=0 RED=3
IR=4 RED=0
IR=4 RED=0
IR=0 RED=3
IR=0 RED=3
IR=2 RED=5
IR=7 RED=0
IR=1 RED=0
IR=0 RED=4
IR=0 RED=7
IR=0 RED=3
IR=5 RED=0
IR=0 RED=0
IR=0 RED=1
IR=0 RED=13
IR=6 RED=0
IR=14 RED=0
IR=0 RED=0
IR=0 RED=0
IR=0 RED=12
IR=11 RED=0
IR=8 RED=0
IR=0 RED=0
IR=0 RED=13
IR=0 RED=8
IR=7 RED=0
IR=7 RED=0
IR=0 RED=0
IR=0 RED=13
IR=1 RED=13
IR=14 RED=0
IR=3 RED=0
IR=0 RED=4
IR=0 RED=13
IR=6 RED=1
IR=17 RED=0
IR=0 RED=0
IR=0 RED=11
IR=0 RED=28
IR=8 RED=13
IR=22 RED=10
IR=3 RED=29
IR=2 RED=56
IR=24 RED=77
IR=46 RED=88
IR=64 RED=118
IR=64 RED=165
IR=88 RED=236
IR=131 RED=311
IR=191 RED=398
IR=260 RED=532
IR=333 RED=720
IR=461 RED=947
IR=614 RED=1213
IR=819 RED=1527
IR=1051 RED=1940
IR=1343 RED=2442
IR=1697 RED=2989
IR=2087 RED=3506
IR=2504 RED=4075
IR=2964 RED=4715
IR=3481 RED=5491
IR=4147 RED=6488
IR=4939 RED=7705
IR=5843 RED=9091
IR=6719 RED=10446
IR=7308 RED=11264
IR=7195 RED=10761
IR=6471 RED=9985
IR=5902 RED=9748
IR=6150 RED=10133
IR=6431 RED=10399
IR=6608 RED=10583
IR=6744 RED=10705
IR=6833 RED=10775
IR=6869 RED=10829
IR=6904 RED=10890
IR=6970 RED=10981
IR=7031 RED=11090
IR=7090 RED=11186
IR=7156 RED=11284
IR=7174 RED=11231
IR=7211 RED=11251
IR=7266 RED=11274
IR=7305 RED=11315
IR=7338 RED=11358
IR=7383 RED=11359
IR=7390 RED=11357
IR=7409 RED=11333
IR=7401 RED=11316
IR=7389 RED=11311
IR=7411 RED=11326
IR=7418 RED=11331
IR=7420 RED=11317
IR=7430 RED=11309
IR=7441 RED=11299
IR=7435 RED=11302
IR=7429 RED=11281
IR=7444 RED=11278
IR=7433 RED=11262
IR=7447 RED=11261
IR=7438 RED=11252
IR=7450 RED=11261
IR=7442 RED=11247
IR=7436 RED=11260
IR=7438 RED=11231
IR=7454 RED=11215
IR=7449 RED=11208
IR=7452 RED=11207
IR=7442 RED=11190
IR=7436 RED=11189
IR=7444 RED=11171
IR=7442 RED=11159
IR=7415 RED=11150
IR=7423 RED=11152
IR=7428 RED=11136
IR=7426 RED=11151
IR=7421 RED=11128
IR=7426 RED=11112
IR=7413 RED=11096
IR=7427 RED=11092
IR=7422 RED=11082
IR=7422 RED=11067
IR=7414 RED=11064
IR=7416 RED=11055
IR=7418 RED=11056
IR=7407 RED=11036
IR=7405 RED=11019
IR=7403 RED=11016
IR=7395 RED=11000
IR=7400 RED=10999
IR=7395 RED=10984
IR=7393 RED=10966
IR=7390 RED=10971
IR=7394 RED=10957
IR=7399 RED=10957
IR=7369 RED=10948
IR=7380 RED=10944
IR=7398 RED=10924
IR=7389 RED=10928
IR=7379 RED=10925
IR=7383 RED=10913
IR=7371 RED=10906
IR=7371 RED=10885
IR=7375 RED=10895
IR=7360 RED=10890
IR=7375 RED=10887
IR=7352 RED=10882
IR=7359 RED=10873
IR=7378 RED=10872
IR=7380 RED=10862
IR=7369 RED=10855
IR=7371 RED=10850
IR=7363 RED=10845
IR=7360 RED=10851
IR=7365 RED=10830
IR=7355 RED=10843
IR=7363 RED=10847
IR=7364 RED=10836
IR=7370 RED=10821
IR=7363 RED=10812
IR=7356 RED=10815
IR=7357 RED=10799
IR=7346 RED=10802
IR=7360 RED=10808
IR=7353 RED=10791
IR=7339 RED=10810
IR=7362 RED=10806
IR=7347 RED=10797
IR=7356 RED=10798
IR=7355 RED=10800
IR=7345 RED=10805
IR=7355 RED=10788
IR=7356 RED=10790
IR=7352 RED=10776
IR=7356 RED=10771
IR=7360 RED=10778
IR=7345 RED=10769
IR=7355 RED=10752
IR=7337 RED=10758
IR=7334 RED=10725
IR=7315 RED=10733
IR=7297 RED=10723
IR=7297 RED=10716
IR=7301 RED=10699
IR=7267 RED=10682
IR=7248 RED=10656
IR=7255 RED=10656
IR=7244 RED=10649
IR=7235 RED=10640
IR=7227 RED=10629
IR=7197 RED=10637
IR=7214 RED=10624
IR=7226 RED=10612
IR=7205 RED=10627
IR=7212 RED=10622
IR=7211 RED=10615
IR=7201 RED=10614
IR=7199 RED=10604
IR=7184 RED=10602
IR=7182 RED=10590
IR=7184 RED=10573
IR=7182 RED=10583
IR=7177 RED=10550
IR=7180 RED=10572
IR=7178 RED=10558
IR=7182 RED=10576
IR=7186 RED=10565
IR=7179 RED=10566
IR=7195 RED=10562
IR=7170 RED=10541
IR=7177 RED=10548
IR=7170 RED=10539
IR=7175 RED=10553
IR=7166 RED=10535
IR=7171 RED=10532
IR=7169 RED=10543
IR=7179 RED=10541
IR=7162 RED=10535
IR=7170 RED=10533
IR=7164 RED=10527
IR=7172 RED=10525
IR=7172 RED=10520
IR=7155 RED=10519
IR=7163 RED=10517
IR=7159 RED=10522
IR=7164 RED=10529
IR=7167 RED=10508
IR=7139 RED=10512
IR=7159 RED=10507
IR=7181 RED=10504
IR=7162 RED=10505
IR=7159 RED=10494
IR=7151 RED=10487
IR=7160 RED=10495
IR=7163 RED=10478
IR=7160 RED=10496
IR=7189 RED=10511
IR=7176 RED=10521
IR=7188 RED=10530
IR=7173 RED=10545
IR=7199 RED=10574
IR=7226 RED=10629
IR=7255 RED=10731
IR=7332 RED=10940
IR=7463 RED=11486
IR=8387 RED=13277
IR=8856 RED=13901
IR=9249 RED=14447
IR=8835 RED=13018
IR=7068 RED=10049
IR=5144 RED=8192
IR=3615 RED=5230
IR=2544 RED=3751
IR=1771 RED=2683
IR=1266 RED=1969
IR=934 RED=1481
IR=709 RED=1144
IR=550 RED=916
IR=448 RED=755
IR=371 RED=624
IR=323 RED=523
IR=272 RED=456
IR=230 RED=399
IR=207 RED=359
IR=196 RED=315
IR=173 RED=279
IR=154 RED=256
IR=133 RED=238
IR=125 RED=212
IR=115 RED=199
IR=116 RED=185
IR=105 RED=174
IR=96 RED=174
IR=100 RED=168
IR=91 RED=146
IR=88 RED=142
IR=81 RED=139
IR=75 RED=143
IR=83 RED=131
IR=81 RED=124
IR=74 RED=111
IR=66 RED=110
IR=60 RED=111
IR=61 RED=100
IR=72 RED=86
IR=59 RED=94
IR=51 RED=96
IR=56 RED=96
IR=58 RED=89
IR=57 RED=85
IR=56 RED=84
IR=43 RED=84
IR=48 RED=93
IR=52 RED=77
IR=55 RED=85
IR=49 RED=85
IR=47 RED=92
IR=54 RED=94
IR=57 RED=84
IR=59 RED=85
IR=54 RED=92
IR=55 RED=102
IR=61 RED=98
IR=69 RED=100
IR=65 RED=101
IR=63 RED=107
IR=67 RED=119
IR=64 RED=115
IR=81 RED=121
IR=76 RED=125
IR=74 RED=134
IR=76 RED=149
IR=85 RED=139
IR=85 RED=126
IR=80 RED=131
IR=67 RED=145
IR=75 RED=146
IR=78 RED=134
IR=89 RED=128
IR=69 RED=135
IR=60 RED=133
IR=65 RED=144
IR=74 RED=119
IR=75 RED=123
IR=58 RED=133
IR=56 RED=142
IR=77 RED=140
IR=85 RED=133
IR=78 RED=136
IR=63 RED=168
IR=71 RED=175
IR=93 RED=186
IR=103 RED=181
IR=101 RED=206
IR=94 RED=235
IR=113 RED=267
IR=137 RED=289
IR=163 RED=297
IR=166 RED=340
IR=170 RED=384
IR=203 RED=423
IR=224 RED=442
IR=241 RED=463
IR=238 RED=486
IR=241 RED=520
IR=254 RED=548
IR=280 RED=552
IR=282 RED=572
IR=280 RED=604
IR=290 RED=648
IR=323 RED=700

### Details of my setup

* Arduino hardware:   Arduino mega 2560
* MAX30100 breakout: could not find anything written on the broad except MAX30100. It is the standard green one similar to 0530.  
* Arduino framework version: 1.8.12
* MAX30100 library version: 1.2.1

MohamedZaky0, 2021-06-21:
@Ardminu  Have u made it work ? How dod u merge it with other sesnsors ?

Ramshask, 2023-05-26:
I'm facing similar issue. Is there any breakthrough to this problem? Help would be appreciated. I am using gy-max30100 with 3 different sensors all interfaced to esp32 but max does not show any values only zeroes. Its working fine when it is seperately interfaced with esp but not with other sensors.