oxullo/Arduino-MAX30100 #80: When call pox.update() in the loop, Arduino don't read Serial data from RX pin

robertomaster, 2020-10-22:
Hello friends of the community. I facing this error a quite long time and I hope you can help me.

When I call the pox.update() method into loop, the Arduino don't receive the entire data from Serial port, who it's connected to a HC-05 bluetooth module. But when I comment it, Arduino receive the data quite well. The variable "palabra" store the data from Serial.

The data of my sensor from the Tester example is like:

Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=34.75C
All test pass.

Press any key to go into sampling loop mode

My setup:
* Arduino hardware: Arduino Pro mini ATMega328P 5V 16MHz
* MAX30100 breakout: serial comunnication issue
* Arduino framework version: 1.8.13
* MAX30100 library version: 1.2.1

My Arduino code is like:

`void loop() {
  if (Serial.available() > 0) {
    char incomingByte = Serial.read();
    if (incomingByte == 'T') {
      palabra = getStringfromProgMem(3);
      c = 0;
    } else {
      palabra += incomingByte;
    }
  }
  
  display.clearDisplay();
  pox.update();
  estado ? (enlace()) : (noenlace());
  unsigned long tiempoActual = millis();
  if (tiempoActual - tiempoAnterior >= intervaloEvento) {
    rate = pox.getHeartRate();
    spo = pox.getSpO2();
    if (rate && palabra != getStringfromProgMem(3)) {
      c++;
    }
    if (c == 60) {
      c = 0;
    }
    Serial.print(rate);
    Serial.print(',');
    Serial.print(spo);
    Serial.print(',');
    Serial.println(c);
    tiempoAnterior = tiempoActual;
  }
  mostrar(rate, 15, 35);
  mostrar(spo, 105, 35);
  mostrar(getStringfromProgMem(0), 10, 52);
  mostrar(getStringfromProgMem(1), 100, 52);

  /*outputValue = (VERF * analogRead(ANALOG)) / 1024;
    smoothedVal = smoothedVal + ((outputValue - smoothedVal) / SAMPLES);
    comparacion(smoothedVal);*/
  //
  display.display();

  estado = digitalRead(STATE);
}`

closed 2020-10-22 by robertomaster

NageshPatil321, 2022-09-27:
hi,
how can you solve this problem and please share me your logic because of this issue i also stuck in my project but i am using MAX30100 sensor instade of HC-05 .
please help me

robertomaster, 2022-09-27:
Hi NageshPatil321,

I send you the functional code that I arrived to run just fine. You must
optimize your code; because of the MAX30100 library size your Arduino is
going to freeze all time (even worse if you use an Arduino UNO or smaller),
and that's the reason for the issue you stock on. I use pointers and other
methods to optimize my code as you're gonna see in my ".ino" file.

*Recommendation:* you must use the interruptions pin of the sensor to write
your code, it's the best approach.

I hope I helped.

Greetings.

Roberto.

thewackyseal, 2022-09-30:
@robertomaster Hi, would you mind sharing a copy of your code with me too? I am having the same issue. Thank you in advance.

robertomaster, 2022-10-02:
Hi *thewackyseal, *sorry for the delay.

Greetings.

Roberto.

thewackyseal, 2022-10-06:
Hi Roberto,

I am not sure where to find your code, I am unable to find it in email or
in github. Would you mind sending it again and mentioning where it is? I am
lost.

Thank you,
thewackyseal

robertomaster, 2022-10-17:
[Tesis.zip](https://github.com/oxullo/Arduino-MAX30100/files/9801493/Tesis.zip)
 Hi, I'm sended you a mail with the code. But I attach the ZIP file with my logic.

Greetings.