oxullo/Arduino-MAX30100 #81: Higher Precision

ChristinaNy, 2021-02-22:
### Troubleshooting checklist

- [YES] I read the README (on master) thoroughly
- [YES] I ran the MAX30100_Tester and I'm going to paste the output down below
- [YES] I filled in all the details of my setup down below

### Description of the issue
I am using the MAX30100_Minimal and it is working fine. However, I would like more decimals (higher precision) of my SpO2 output. Can anyone help with this regarding the pox.getSpO2() function? 
I tried using a float:

Serial.print(float(pox.getSpO2()),4);

But it just put four zeroes after my SpO2 result (like 98.0000 %). What I am looking for is to get numbers like 97.5482 %.
Hope someone can help. Thank you.

### Output from MAX30100_Tester example

### Details of my setup

* Arduino hardware: Arduino UNO
* MAX30100 breakout:
* Arduino framework version:
* MAX30100 library version: MAX30100_Minimal

selman-nus, 2021-03-10:
Hi, you may need to change how the getspo2 function defined in the library files. Probably it is an integer and you need to change it to float. However, even if this solves your problem by showing decimals, it doesn't mean it is more precise really. Even medical oximeters don't show decimals. So what you will be getting is just noise. I won't suggest you rely on that data for your project.

ChristinaNy, 2021-03-10:
Thank you for your help. That makes sense.
I decided to use the RAW script and calculate the values manually in Excel.

selman-nus, 2021-03-10:
are you going to confirm your results using a medical oximeter?

ChristinaNy, 2021-03-10:
Yes I have done that. The commercial pulseox gives 99 %SpO2 and my calculations give an average of 98.2 %Sp02. Based on the uncertainties in both I conclude that these are statistically similar.
These are the formulas I use:
R = (AC of Red / DC of Red) / (AC of IR / DC of IR)
% SpO2 = 110 – 25 × R

selman-nus, 2021-03-10:
Commercial ones generally use moving average over spo2 results to smooth the output values. So it is not like an instantaneous reading result, just be careful with that