oxullo/Arduino-MAX30100 #83: MAX30100 gives zero outputs

ZahraaSabah97, 2021-05-19:
### Troubleshooting checklist

- [done] I read the README (on master) thoroughly
- [done] I ran the MAX30100_Tester and I'm going to paste the output down below
- [done] I filled in all the details of my setup down below

### Description of the issue
MAX30100 works fine in a simple example, but it gives zero outputs in a project.
my graduation project collects the patient's facial signs (heart rate, SpO2, Temp) and sends them via ESP8266 to Thingspeak, but I don't know why max30100 gives zero outputs and doesn't send the results to thingspeak?

Here is my code:
```

#define USE_ARDUINO_INTERRUPTS true    
#define DEBUG true
#define SSID "Mohammed"           // "SSID-WiFiname"
#define PASS "1234567890abc"      // "password"
#define IP "184.106.153.149"      // thingspeak.com ip
#define REPORTING_PERIOD_MS     1000
#include <Event.h>
#include <Timer.h>
#include <Wire.h>
#include "MAX30100_PulseOximeter.h"
#include <SoftwareSerial.h>
#include "Timer.h"
Timer t;

PulseOximeter pox;
uint32_t tsLastReport = 0;

String msg = "GET /update?key=ETIEQG6HXQYK2TWG";    //API key
SoftwareSerial esp8266(2,3);    //RX, TX

//Variables
float myTemp;
int myBPM;
int mySpO2;
String BPM;
String temp;
int error;
int raw_myTemp;
float Voltage;
float tempC;

void onBeatDetected()
{
    Serial.println("Beat!");
}

void setup()
{
    Serial.begin(9600); 
    esp8266.begin(115200);
    Serial.print("Initializing pulse oximeter..");

    if (!pox.begin()) {
        Serial.println("FAILED");
        for(;;);
    } else {
        Serial.println("SUCCESS");
    }
    pox.setOnBeatDetectedCallback(onBeatDetected);
    
    Serial.println("AT");
    esp8266.println("AT");
    delay(3000);        
      
    if(esp8266.find("OK"))
    {
      connectWiFi();
    }
     t.every(10000, getReadings);
     t.every(10000, updateInfo);
}

void loop()
{
   pox.update();
   myBPM = pox.getHeartRate();
   mySpO2 = pox.getSpO2();
    if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
        Serial.print("Heart rate: ");
        Serial.print(myBPM);
        Serial.print("bpm / SpO2: ");
        Serial.print(mySpO2);
        Serial.println("%");

        tsLastReport = millis();
}
 start: //label
 error=0;
   t.update();    //Resend if transmission is not completed
 if (error==1)
  {
  goto start;     //go to label "start"
  } 
  delay(4000);      
}

void updateInfo()
{
  String cmd = "AT+CIPSTART=\"TCP\",\"";
  cmd += IP;
  cmd += "\",80";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(2000);
 
  if(esp8266.find("Error"))
  {
    return;
  }
  cmd = msg ;
  cmd += "&field1=";    //field 1 for BPM
  cmd += BPM;
  cmd += "&field2=";  //field 2 for temperature
  cmd += temp;
  cmd += "\r\n";
  Serial.print("AT+CIPSEND=");
  esp8266.print("AT+CIPSEND=");
  Serial.println(cmd.length());
  esp8266.println(cmd.length());
 
  if(esp8266.find(">"))
  {
    Serial.print(cmd);
    esp8266.print(cmd);
  }
  else
  {
    Serial.println("AT+CIPCLOSE");
    esp8266.println("AT+CIPCLOSE");
    //Resend...
    error=1;
  }
}

boolean connectWiFi()
{
  Serial.println("AT+CWMODE=1");
  esp8266.println("AT+CWMODE=1");
  delay(2000);
  String cmd="AT+CWJAP=\"";
  cmd+=SSID;
  cmd+="\",\"";
  cmd+=PASS;
  cmd+="\"";
  Serial.println(cmd);
  esp8266.println(cmd);
  delay(5000);
  
  if(esp8266.find("OK"))
  {
    return true;
  }
  else
  {
    return false;
  }
}

void getReadings()
{
  raw_myTemp = analogRead(A1);
  Voltage = (raw_myTemp / 1023.0) * 5000;       // 5000 to get millivots.
  tempC = Voltage * 0.1; 
  myTemp = (tempC * 1.8) + 32;                  // conver to F
  Serial.print("TEMP= ");
  Serial.println(myTemp);
  myBPM= pox.getHeartRate();
    delay(20);            
    char buffer1[10];
    char buffer2[10];
    BPM  = dtostrf(myBPM, 4, 1, buffer1);
    temp = dtostrf(myTemp, 4, 1, buffer2);  
  }
```
### Output from MAX30100_Tester example
[image]

### Details of my setup

* Arduino hardware: Arduino UNO
* MAX30100 breakout: https://www.tokopedia.com/wdelectronic/max30100-heart-rate-sensor-module-green-board?src=topads
* Arduino framework version: Arduino IDE 1.8.15
* MAX30100 library version:

geekypaulson, 2021-05-24:
It's because of the sampling frequency, which is set to 100Hz. The creator we can't use MAX30100 with Thingspeak because there is delay when we update on Thingspeak. This causes the algorithm to average consecutive values or jitter. You can refer to #49

geekypaulson, 2021-05-24:
You can also check #52 and #54