oxullo/Arduino-MAX30100 #84: I get value of Zero if i merged my code with any other sensor or tried to upload it on Ubidots

MohamedZaky0, 2021-06-21:
### Troubleshooting checklist

- [Yes] I read the README (on master) thoroughly
- [Yes] I ran the MAX30100_Tester and I'm going to paste the output down below
- [Yes] I filled in all the details of my setup down below

### Description of the issue
I can't merge MAX30100 code with any other sensors i always get 0s
### Output from MAX30100_Tester example
[image]

### Details of my setup
The issue is kinda complicated as MAX30100 had been modified by desoldering the 3 resistors 4.7k and making external resistors .. I have been trying to make MAX30100 works with another codes but with no use on Nodemcu v2 / Arduino Uno .. tried to just upload the data on Ubidots but i get zeros always .. i even tried to use Arduino as master and send data through Serial communication but no use i get zeros value .. I traced the code all along on the MAX30100 code alone if i added delay more than 90 ms it will not detect any beat ... So is there anyway i can merge MAX30100 with ECG Sensor AD8232 and upload the data to Ubidots ?

One of my trial codes: 
#include <PubSubClient.h>

#include <ESP8266WiFi.h>

/****************************************
 * Include Libraries
 ****************************************/
#include "UbidotsESPMQTT.h"
#include <Wire.h>
#include "MAX30100_PulseOximeter.h"

/****************************************
 * Define Constants
 ****************************************/
#define TOKEN "BBFF-19b5dcYiCZyQuGqI9TUynsDhSNYJEa" // Your Ubidots TOKEN
#define WIFINAME "Dido" //Your SSID
#define WIFIPASS "Hidden" // Your Wifi Pass
 
#define REPORTING_PERIOD_MS     1000
 
PulseOximeter pox;
uint32_t tsLastReport = 0;

int hr = 0;
int spo= 0;
Ubidots client(TOKEN);

/****************************************
 * Auxiliar Functions
 ****************************************/

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

void onBeatDetected()
{
    Serial.println("Beat!");
}
/****************************************
 * Main Functions
 ****************************************/

void setup() {
  // put your setup code here, to run once:
  client.ubidotsSetBroker("industrial.api.ubidots.com"); // Sets the broker properly for the industrial account
  client.setDebug(true); // Pass a true or false bool value to activate debug messages
  Serial.begin(115200);
  client.wifiConnection(WIFINAME, WIFIPASS);
  client.begin(callback);
  Serial.print("Initializing Connection..");
  delay(2000);
  Serial.print("Initializing pulse oximeter..");
    pinMode(16,OUTPUT);
    // Initialize the PulseOximeter instance
    // Failures are generally due to an improper I2C wiring, missing power supply
    // or wrong target chip
    if (!pox.begin()) {
        Serial.println("FAILED");
        for(;;);
    } else {
        Serial.println("SUCCESS");
    }
     pox.setIRLedCurrent(MAX30100_LED_CURR_7_6MA);
 
    // Register a callback for the beat detection
    pox.setOnBeatDetectedCallback(onBeatDetected);
  }

void loop() {

 // Serial.print("Loop Started");
  // put your main code here, to run repeatedly:
    if(!client.connected()){
      client.reconnect();
      }
     pox.update();
    if (millis() - tsLastReport > REPORTING_PERIOD_MS) {
        Serial.print("Heart rate:");
        hr = pox.getHeartRate();
        Serial.print(hr);
        Serial.print("bpm / SpO2:");
        spo=pox.getSpO2();
        Serial.print(spo);
        Serial.println("%");
        client.add("Heart rate",hr);
        client.add("Oximeter",spo);
        client.ubidotsPublish("Er7amny");
        client.loop();
        tsLastReport = millis();
    }

}

* Arduino hardware: Nodemcu v2
* MAX30100 breakout:  The Green one with hardware issue
* Arduino framework version: 1.8.15
* MAX30100 library version: latest

gatorkoco, 2021-06-24:
https://www.teachmemicro.com/max30100-arduino-heart-rate-sensor/

MohamedZaky0, 2021-08-28:
> please help me if issue is resolved , i am also facing the same issue.

You need to make the modification of resistors thing in the sensor and it
will work fine

Banadesh708, 2021-08-28:
Shall I have to vary the resistance or change the circuit connections , can u please detail me out , I have been stuck in this from past 3 days.

Praadnya, 2023-02-28:
using setClock function of Wire library ensure to maintain the clock speed to be 100000 in the setup function