oxullo/Arduino-MAX30100 #91: Big values for GY-MAX30100

PaulaStef, 2023-05-16:
### Troubleshooting checklist

- [ ] I read the README (on master) thoroughly
- [ ] I ran the MAX30100_Tester and I'm going to paste the output down below
- [ ] I filled in all the details of my setup down below

The GY-MAX30100 sensor gives wrong values for heart rate - the values are mostly to big, sometimes it gives non-zero values even when I'm not touching the sensor. I changed the IR led current to pox.setIRLedCurrent(MAX30100_LED_CURR_7_6MA) and this way it gives 0 when I dont touch the sensor but it still gives large value for HR.

Tester Output:

Initializing MAX30100..Success
Enabling HR/SPO2 mode..done.
Configuring LEDs biases to 50mA..done.
Lowering the current to 7.6mA..done.
Shutting down..done.
Resuming normal operation..done.
Sampling die temperature..done, temp=0.00C
WARNING: Temperature probe reported an odd value

Press any key to go into sampling loop mode

I connected the sensor to an Arduino MEGA 2560
Vin- 3.3V
GND-GND
SCL-SCL
SDA-SDA

Output :
Heart rate:119.47bpm / SpO2:95%
Beat!
Beat!
Heart rate:144.82bpm / SpO2:96%
Beat!
Beat!
Beat!
Beat!
Heart rate:186.02bpm / SpO2:95%
Beat!
Beat!
Beat!
Beat!
Heart rate:227.04bpm / SpO2:94%
Beat!

Arduino IDE version 2.0.3
I got the zip archive from the master branch and added to arduino library

this is the sensor I used - https://www.sigmanortec.ro/senzor-puls-optic-gy-max30102?gclid=CjwKCAjw04yjBhApEiwAJcvNoe6ep7EQXLRGdOR0HxnFfKtAWjC-iBOqamjsQUAgWZ4-ppDLwpj9ohoChpAQAvD_BwE

Guiaochino, 2023-06-11:
I am also having this problem with my GY-MAX30100. Is there any way that we can fix this?

PaulaStef, 2023-06-16:
@Guiaochino I managed to get the values a little lower by changing some constants in MAX30100_BeatDetector.h file like BEATDETECTOR_MAX_THRESHOLD and  BEATDETECTOR_MIN_THRESHOLD. I would still get some spikes so I used a median filter to balance them. Not the perfect solution but I do get better values now.

BloggingKIng, 2023-06-24:
Hi did any one of you solved this problem? I am facing the same problem

PaulaStef, 2023-06-25:
@BloggingKIng I manage to reduce the values a little by changing some values from MAX30100_beatDetector.h. I changed a bit the threshold values and it did lower the values. I believe that it would also help to change the frequency values from butterworth filter, although I did not try that. 
I still get some spikes now and then but I used a median filter and is working decent enough.

BloggingKIng, 2023-06-25:
@PaulaStef thanks for your reply. My problem is that actually Max30100 sends random values and some time eventually stops sending values. I will then have to reset Arduino Uno to get it working back again. 

I am interested in that median filter thing. Can you tell  m, how did you implement it?

PaulaStef, 2023-06-26:
@BloggingKIng I stored 5 consecutive values of pulse from the sensor in an array, sorted it and then a used the value in the middle instead of the current value that the sensor gave (at index 2 starting from 0). When you get a new value you replace it with an existing one in the array. This filter is usually used to reduce noise in images but it worked pretty well.